AprilRobotics/apriltag #3: problem about rotate90

lishuaijielsj, 2018-12-28:
hi, 
   i want to consult that whether the rotate90 in this version has the same function as rotate90 in version2? because  i test it on Tag36h11,  and  i find that the version3 do not work well when the Apriltag rotate 90 degree.
### **rotate90 in version2 is:**
static uint64_t rotate90(uint64_t w, uint32_t d)
{
    uint64_t wr = 0;
    for (int32_t r = d-1; r >=0; r--) {
        for (int32_t c = 0; c < d; c++) {
            int32_t b = r + d*c;
            wr = wr << 1;
            if ((w & (((uint64_t) 1) << b))!=0)
                wr |= 1;
        }
    }

    return wr;
}
### **code in version3 is:**
static uint64_t rotate90(uint64_t w, int numBits)
{
    int p = numBits;
    uint64_t l = 0;
    if (numBits % 4 == 1) {
	p = numBits - 1;
	l = 1;
    }
    w = ((w >> l) << (p/4 + l)) | (w >> (3 * p/ 4 + l) << l) | (w & l);
    w &= ((1L << numBits) - 1);
    return w;
}

mkrogius, 2019-01-03:
Thank you for reporting this issue and especially for submitting the pull request with the fix. I have confirmed that it works and merged the fix.

closed 2019-01-03 by mkrogius

lishuaijielsj, 2019-01-07:
> Thank you for reporting this issue and especially for submitting the pull request with the fix. I have confirmed that it works and merged the fix.

Thanks for your response

commit 2a0e3ac9 (2019-01-03): Fixrotate90 (#4) fixes #3

Fix sign of rotation.