AprilRobotics/apriltag #7: Why maxiters equals 1?

Ivy-Tang, 2019-01-19:
Hi! 

Firstly, thank you for your great work! 
I'm writing here because I had a problem. I ran the project, but it only printed three lines and quit. The three lines are shown below:
'Summary
hamm 0 0 0 0 0 0 0 0 0 0 0.000
0'
It seems the parameter 'maxiters' equals one, but I don't know how to fix this. Would you help me with this one? Thank you in advance.

Best,
Ivy

allsey87, 2019-01-19:
I think you need to add a bit more information regarding exactly how to recreate this issue. For example, are your images coming from a camera or from a file. If you use a file, the issue would be much easier to recreate and would eliminate any potential lighting / camera parameter issues.

mkrogius, 2019-01-28:
Closing for inactivity

closed 2019-01-28 by mkrogius