AprilRobotics/apriltag #13: ROS node

leifole, 2019-03-08:
Hi,
seems there are multiple repos trying to wrap apritags for ROS, but most of them seem outdated. Are you planning on releasing a ROS node and keeping it up to date? Would be awesome...

wxmerkt, 2019-03-08:
Hi @leifole - FYI: We are in the process of updating/upgrading https://github.com/dmalyuta/apriltags2_ros/ with the aim of releasing it via the buildfarm as binaries. Would love to see an official ROS node (and we'd adjust our planning accordingly). Is there a feature missing from the above?

mkrogius, 2019-03-08:
Hi, 
I am not planning on writing an official ROS node, mostly because we don't use ROS internally so it would be extra effort to test and maintain it. I am glad that @wxmerkt has a wrapper for ROS since I think that this makes AprilTags much more accessible (and I'm very glad to hear it is being updated to AprilTags3!).

rgreid, 2019-04-07:
Linking to a discussion here https://github.com/dmalyuta/apriltags2_ros/issues/45, which is discussing merging Apriltag 3 into a repo/package called `apriltag_ros` and based on `apriltags2_ros`.  As @leifole pointed out, the other ROS wrappers appear outdated.

rgreid, 2019-05-20:
Quick update on this one: The `apriltag_ros` package/repo is now in the official April Robotics org, available here: https://github.com/AprilRobotics/apriltag_ros   

@mkrogius you can probably close this issue.

leifole, 2019-05-20:
Nice! Thanks for the effort.

closed 2019-05-24 by mkrogius