AprilRobotics/apriltag #14: Add/Invent a version and tag latest stable code

milijan, 2019-03-08:
It would be useful to adopt a version scheme and tag the code accordingly. This is try for other AprilRobotics repositories. In particular, this would enable some level of traceability for corporate users.

mkrogius, 2019-03-08:
Hi, can you go into a bit more detail about what exactly you want? Master should be latest stable code at all times.

atuleu, 2019-04-02:
👍 I think itbwould be nice to install a semantic versionning of apriltag binaries. More information would be available here : https://semver.org/

Ensuring that the master branch will always maintain stable release is sometimes not enough. If you open your library for other to include in their own projects, asking them to target the master branch, these users will often require to be able to produce repeatable builds of their project, and therefore need to target a specific revision. And they would manually target a new version for example when an upstream issue is fixed. This workflow allows for hot patch fixing (when for example a patch is needed in their project but is not already merged upstream).

Of course you could just ask to target a specific commit hash, but it is common usage for people to look for a tag in the form vX.Y.Z. For example you could choose that the current commit in the repo is the official v3.0.0 release, and increment version when releasing following the guidelines in https://semver.org/

mkrogius, 2019-04-05:
Alright, I'm adding a version number/release to the project.

closed 2019-04-08 by mkrogius

milijan, 2019-04-08:
> Alright, I'm adding a version number/release to the project.

@mkrogius really sorry for the late reply. I saw your request but never got round to comment. @atuleu answer great.