AprilRobotics/apriltag #15: Homography NULL pointer under ARM architecture

AleksandarPetrov, 2019-03-13:
With @surirohit we have been working on [these](https://github.com/duckietown/apriltags3-py) Python bindings for your Apriltags3 library. We are building these for both x86 and arm architectures. We experience no issues with x86. However under arm (on a RPi), we get the following error when we try to access the homography of a detected April Tag:

```
Traceback (most recent call last):
 File "apriltags3.py", line 465, in <module>
   tags = at_detector.detect(img, True, camera_params, parameters['sample_test']['tag_size'])
 File "apriltags3.py", line 360, in detect
   homography =_matd_get_array(tag.H).copy()
 File "apriltags3.py", line 124, in _matd_get_array
   mat_ptr.contents.data,
ValueError: NULL pointer access
Segmentation fault
```
The exact same code runs with no issue under Ubuntu on a standard Laptop PC with a 64bit Intel x86 processor.

Given that the issue is architecture-depended (x86 vs arm, 64bit vs 32bit) we suspect that this could potentially be an issue on your end that you might be interested in looking into.

allsey87, 2019-03-14:
I have seen an issue that is possibly related to this before. When trying to get the pose of an Apriltag using OpenCV's solvePnP (CV_ITERATIVE method) on an OMAP4 board, one orientation of the tag would cause the method to return NaN. This wasn't the case when running the code under Ubuntu on a 64-bit Intel machine. My code used C language implementation from back in 2015 and then fed the corner coordinates of the tag into OpenCV.

mkrogius, 2019-03-14:
Thanks for reporting this issue. I have tried running the code on a RPi, but for me the homography pointer exists and the tags are detected. Do you run into this issue for all or just some images? Does the code print out any errors?

@allsey87 I'll take this opportunity to plug the pose estimation code we wrote as part of AprilTag 3. I think it should do better than OpenCV for "hard" orientations, but I haven't done a comparison yet.

AleksandarPetrov, 2019-03-29:
@mkrogius It failed for different images and the only error we were getting is in the original post above.

mkrogius, 2019-03-29:
I haven't tested this on a RPi but it looks like the issue is probably that the apriltag_detection struct is defined differently in the python vs c code. In AprilTags 3 we removed the "goodness" parameter but it is still present in the python definition.

I tested this on x86 and it results in the value for the "decision_margin" being read as the (now non-existent) "goodness" in the python code while the "decision_margin" is read as zero. On x86 the compiler adds enough padding such that the homography is still at the same memory location regardless of whether the goodness parameter exists. I bet that on ARM it ends up in a different location.

AleksandarPetrov, 2019-03-29:
@mkrogius This sounds like a very likely culprit. Thank you for the information. I will try to fix it and test it these days.

closed 2019-05-24 by mkrogius