AprilRobotics/apriltag #16: Release to ROS1 and ROS2

christianrauch, 2019-03-14:
Following the integration of CMake and the `package.xml` metadata in https://github.com/AprilRobotics/apriltags/pull/11, this issue is for tracking the release of the library to the official ROS1 and ROS2 repo.

I released to [ROS2 crystal](https://github.com/ros/rosdistro/blob/09c3906fa3121ca8351108c1940d3e518ec9305d/crystal/distribution.yaml#L148) a while ago from my now obsolet repo. A future PR to `rosdistro` would need to replace this section.

rgreid, 2019-04-05:
@christianrauch are you planning to do this PR? I have `master` building fine on 18.04/Melodic.

christianrauch, 2019-04-05:
@rgreid I can do the release. However, I still prefer that the release repository is placed in a public visible place, e.g. `https://github.com/AprilRobotics/apriltags-release` or `https://github.com/ros-gbp-release` with access by multiple maintainers to prevent that the package gets orphaned in the future.

mkrogius, 2019-04-05:
Which of those two places would you prefer to host the release repository? I would have a slight preference for hosting in ros-gbp, but if that is more work to set up than I would be happy to make a new repo in the AprilRobotics org.

christianrauch, 2019-04-05:
I've no preferences. In any case, I would need to get access to that organisation or release repo.

If I recall correctly, the release process requires that the upstream repo has tags that match the version in the `package.xml` and that the version increments according to the [Semantic Versioning](https://semver.org/).
I originally stated the `BSD` license. Is this still correct?
Can you update the version, and if necessary the license, in the `package.xml`?

wxmerkt, 2019-04-05:
Looping in @paulbovbel regarding a `ros-gbp` release repository.

rgreid, 2019-04-07:
To help others, since I had to google it, the requirements for a ROS 'release repo' are [here](http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease). 

@mkrogius one thought might be to rename this repo to `apriltag`, and then transfer/fork the `dmalyuta/apriltags2_ros` node repo here, so that we have two repos named consistently:
 - https://github.com/AprilRobotics/apriltag  
 - https://github.com/AprilRobotics/apriltag_ros 

As discussed [here](https://github.com/dmalyuta/apriltags2_ros/issues/45). Just my two cents!

mkrogius, 2019-04-08:
I'm happy to rename it. I just need to check there aren't negative effects of renaming a repo

rgreid, 2019-04-11:
@mkrogius Github automatically creates redirects so both old and new URLs continue to work.

mkrogius, 2019-04-11:
I've gone ahead and renamed this repo to apriltag

wxmerkt, 2019-04-11:
Fantastic, thank you. Regarding release repository, @paulbovbel - is it possible to get one within @ros-gbp?

paulbovbel, 2019-04-12:
I haven't heard back from the org owners whether that would be appropriate. It probably just makes sense to make an apriltag-release repo in this org, that way the permissions are under your control.

wxmerkt, 2019-04-17:
@mkrogius can you please create a apriltag-release repository here as well if that's okay so the library can be released into ROS? Thank you :)

mkrogius, 2019-04-19:
Done, and I've added the same collaborators as for apriltag_ros

rgreid, 2019-04-19:
Great. Do we need an `apriltag_ros-release` repo also?

wxmerkt, 2019-04-19:
Yes, good call @rgreid.

wxmerkt, 2019-05-13:
Release requests for kinetic and melodic are pending:

- Kinetic: https://github.com/ros/rosdistro/pull/21221
- Melodic: https://github.com/ros/rosdistro/pull/21220

christianrauch, 2019-05-14:
@wxmerkt Could you also do a release for ROS dashing?
I would do a dashing release if I get access to the release repo. However, for not having to do yet another patch release just for adding me as maintainer, only you would be notified on dashing release failures. This may not be a problem, since this repo does not depend on ROS packages and if a release on melodic succeeds, it should also succeed for dashing.

wxmerkt, 2019-05-14:
@christianrauch done - https://github.com/ros/rosdistro/pull/21226

rgreid, 2019-05-21:
On a related note, should we push `apriltag_ros` also?  The old `apriltags2_ros` package is currently in `rosdistro`:  https://github.com/ros/rosdistro/blob/master/kinetic/distribution.yaml#L236

wxmerkt, 2019-05-21:
Yes, I was waiting for these pull requests to clear. I am happy to bloom `apriltag_ros` once we have a `apriltag_ros-release` repository.

mkrogius, 2019-05-24:
I created the apriltag_ros-release repo

wxmerkt, 2019-05-25:
Pull requests for apriltag_ros to kinetic and melodic are open - note: I haven't removed the original apriltags2_ros release in kinetic.

Kinetic: https://github.com/ros/rosdistro/pull/21348
Melodic: https://github.com/ros/rosdistro/pull/21349

wxmerkt, 2019-06-03:
The release to Kinetic just got reverted amidst build failures. Unfortunately, the buildfarm has some hiccups and I cannot access detailed logs right now - cf. https://github.com/ros/rosdistro/pull/21437

@mkrogius Have you build on ARM before and have you seen the long buildtimes they advise? I don't understand the long buildtimes on amd64, apart from maybe memory limitations.

mkrogius, 2019-06-03:
@wxmerkt I have come across this issue before with long build times. It seems that anything -O2 and above causes gcc on ARM to take a super long time to build the larger tag families. In theory this should have been fixed with the __attribute__((optimize("O0"))) on the tag_*_create functions.

I don't know if this fix was ever verified, since for me, the build times are pretty reasonable on my raspberry pi. I would need to spend some more time, maybe later this week to look into this issue more. For now, one option would be to not build the 3 large tag families by default.

 Some references:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=921256
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=920894
For this debian package, the solution by Dima Kogan was to load the tag codes at runtime instead of having them compiled into the binary.

wxmerkt, 2019-06-03:
Thank you, Max - this are great pointers. Is apriltag now part of a debian release? And has the loading-at-runtime fix been upstreamed into this repository?

mkrogius, 2019-06-03:
I think we might be part of a debian release now, yes. The loading-at-runtime fix was never upstreamed, but that was because we wanted a different implementation of loading at runtime. Since we control the generation of these files, it makes more sense to just directly generate the files to be loaded.

I also think its worth it to try compiling these files with O0, to see if that fixes it (if we had a way to reproduce the issue)

mkrogius, 2019-06-13:
I've pushed a fix for the compile time issue to the AprilTag repo. The issue was that on older (5.4) versions of gcc, track-vars is very slow: https://stackoverflow.com/questions/2954473/variable-tracking-is-eating-my-compile-time.

I made a pretty minimal change to fix this, instead of specifying "O0" in the __attribute__, I specify it for the whole file in the build rule (only for the tag*.c files where performance doesn't really matter).

wxmerkt, 2019-06-13:
Thank you for tracking this down and fixing it. Do you mind updating the `package.xml` version tag to `3.1.1` and adding a new git tag `3.1.1`? Then we can do another round of releases for ROS :)

mkrogius, 2019-06-13:
No problem: https://github.com/AprilRobotics/apriltag/releases/tag/3.1.1

wxmerkt, 2019-06-14:
Thank you - I have made the new release requests.

wxmerkt, 2019-10-07:
@mkrogius we can close this issue :-)

closed 2019-10-07 by christianrauch