AprilRobotics/apriltag #19: Pose estimation only on 4 points ??

AKEOPLUS-boris-bocquet, 2019-03-28:
Dear AprilRobotics teams,

I have tried your library and I could compile the codes and run on my images on first try with no efforts in almost no time. Many thanks for that. Then I focused on pose estimation. Reading at the code, it looks like estimate_tag_pose calls estimate_tag_pose_orthogonal_iteration,  which only uses the 4 corners of the tag (orthogonal_iteration function is called on v and p with n_points == 4). Am I correct ? Why only on the 4 corners ? Don't you have all the pixel values of each corners of the landmark ? And the equivalent coordinate in the real world (scaled using the given tagsize) ? If so you could provide all px / real word correspondances to your PnP solver (or equivalent) to get a more robust pose estimation. 
What's your opinion ? 

Best Regards
Boris Bocquet

mkrogius, 2019-03-29:
Hi,

I'm glad it was easy to get up and running with the library!

You are correct that we only use the 4 corner points for PnP, this is because it gives us very similar accuracy as using full knowledge of the tag.

I had the same idea of using all of the points on the tag for PnP and I experimented with this idea last summer. My approach was to project an image of the tag into the camera image using an estimate of the tag pose, compute how well the projection matched the actual image, and then improve the estimate via coordinate ascent. This approach worked in that it gave similar tag poses as the PnP solution with 4 points, but it did not give any additional accuracy on the dataset that I was using. It is possible that the dataset I was using did not have super accurate ground truth, but I was convinced enough that no accuracy improvements were possible that I moved on to exploring other ideas for improving AprilTags.

Best,
Max

AKEOPLUS-boris-bocquet, 2019-04-11:
Dear Max, 

thank you for your quick reply. It took me some times to think about it, and I was really busy those days as well. Your tests are good news, and they show that your are wise. I may have some doubts anyhow. But there are also due to the fact that I don't know how your algorithm work. My first doubt will be that you use only 4 points, which is the minimum required. Are these 4 points trustable? Are they optimized? What if they are approximatively located on the image : not finely located but your landmark identification can handle it ? Second doubt : you don't compensate the lense distorsion. And maybe this is why you saw no accuracy difference with your test on all corners and only 4 corners. This point is less important than the first one (since I am not expecting to calibrate precisely my camera in my application). 

Then, since you also perform a pose estimation on 4 points, may I ask you to comment the pros and cons of your library against a pose estimation on a QRcode corners, just like they do in  [VISP pose on QRcode (using Zbar)](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-pose-estimation-qrcode.html) .

Let me thank you in advance.

Best Regards.
Boris

mkrogius, 2019-04-11:
Hi Boris,

The 4 points that are used should be quite accurate since they are localized by using the entire detection border of the tag, which consists of large black-white borders. My tests did compensate for lens distortion (by running the detector on a rectified image). 

In your link it looks like they are also using only 4 points. In that specific example it looks like they use a linear pose solver (although they also have nonlinear solvers). The apriltag library uses a nonlinear solver when you call estimate_tag_pose. I would expect that both the apriltag and visp nonlinear solvers have similar accuracy, so any difference in accuracy would come from how accurate the detections of the 4 points are.

Best,
Max

AKEOPLUS-boris-bocquet, 2019-04-19:
Hi Max, 
many thanks for your valuable answers. I think I will have to dig into the zbar QRcode corner extraction then. Anyhow, it seems that there are no bad alternatives and both are worth testing on the field.

All the best,
Boris

closed 2019-04-19 by AKEOPLUS-boris-bocquet

pra-dan, 2021-01-20:
Hi @mkrogius and @AKEOPLUS-boris-bocquet thanks for this informative issue. 👍 

I have a doubt, please help. I found that the [PnP solver provided by OpenCV](https://docs.opencv.org/master/d7/d53/tutorial_py_pose.html) uses a simple chessboard pattern to successfully demonstrate pose detection. The *only* advantage AprilTag (or any other popular feducial marker system) provides is that the 4-edge-detection is robust than the chessboard, which looks identical when rotated in multiples of 90 degrees. Another benefit they offer is an encoded id/number which can be used to label objects.

Please correct me if wrong. This will immensely help me. :)

mkrogius, 2021-01-20:
AprilTag (or other fiducials) tend to be used for very different purposes than a checkerboard pattern. Fiducials are used to track objects or localize a robot. Checkerboard patterns are typically only used for camera calibration.

Here are some facts about fiducial vs checkerboard
Fiducial:
- 4 detected points
- Can be detected from further away
- Many unique tags
- cannot be detected through occlusions

Checkerboard:
- many detected points
- some occlusion resistance
- Z-rotation ambiguous in increments of 90 degrees
- Few unique checkerboards

And here are some untested guesses
Fiducial:
- May have higher recall?
- May have higher precision?
- May have faster detection?
Checkerboard:
- May have slightly more accurate pose estimation?