AprilRobotics/apriltag #22: Error, more than one new minima found.

jkgarbe, 2019-04-02:
Hello, 
I edited the opencv_demo by adding the following because I am interested in pose:

`               //POSE DETECTION:
                // First create an apriltag_detection_info_t struct using your known parameters:
                apriltag_detection_info_t info;
                info.det = det;
                info.tagsize = 100;
                info.fx = 630;
                info.fy = 630;
                info.cx = 640;
                info.cy = 360;
                // Then call estimate_tag_pose.
                apriltag_pose_t pose;
                //There is an error with the following line:
                double err = estimate_tag_pose(&info, &pose);
                cout << "Err: " << err <<endl;`

When I execute opencv_demo, it works for about 5 seconds, then crashes with this error:

Error, more than one new minima found.
*** Error in `./opencv_demo': free(): invalid pointer: 0x00007f01cd5ad81f ***
======= Backtrace: =========

Then it lists a bunch of locations.
How can I fix this?

mkrogius, 2019-04-05:
Hi,

1) Does opencv_demo work with no edits made?
2) Can you please post the full output? I don't believe the "Error, more than one new minima found." is related to the crash.

Best,
Max

jkgarbe, 2019-04-08:
@s-trinh Thank you! My issue is fixed.

PR #23 (2019-04-08): Check for NULL before calling matd_destroy
`pose2.R` is initialized only if there is one new minima:
https://github.com/AprilRobotics/apriltags/blob/a5ab81dd79fd294b77940efad889a820509e83ae/apriltag_pose.c#L412-L426

`pose2.t` is initialized only if `pose2.R` is not `NULL`:
https://github.com/AprilRobotics/apriltags/blob/a5ab81dd79fd294b77940efad889a820509e83ae/apriltag_pose.c#L496-L502

Probably should fix #22
closed 2019-04-08 by mkrogius