AprilRobotics/apriltag #24: Invalid Pointer Error

ZhouYixuanRobtic, 2019-04-08:
Thanks for your excellent works, whereas I have some questions waiting your generous answers.
- **What's the unit of tagsize**
 The tagsize parameter should give the physic size of tag, then what's its unit. Meters or millimeter? for example, should I write my tagsize as 0.178 or 178 for a real 178 mm X 178 mm tag. 
- **Invalid Pointer**
I write my c++ script as below, and once I enable my pose_estimation,I got Error.
 ```c++
for (int i = 0; i < zarray_size(detections); i++)
        {
            apriltag_detection_t *det;
            zarray_get(detections, i, &det);
            line(frame, Point(det->p[0][0], det->p[0][1]),
                 Point(det->p[1][0], det->p[1][1]),
                 Scalar(0, 0xff, 0), 2);
            line(frame, Point(det->p[0][0], det->p[0][1]),
                 Point(det->p[3][0], det->p[3][1]),
                 Scalar(0, 0, 0xff), 2);
            line(frame, Point(det->p[1][0], det->p[1][1]),
                 Point(det->p[2][0], det->p[2][1]),
                 Scalar(0xff, 0, 0), 2);
            line(frame, Point(det->p[2][0], det->p[2][1]),
                 Point(det->p[3][0], det->p[3][1]),
                 Scalar(0xff, 0, 0), 2);

            stringstream ss;
            ss << det->id;
            String text = ss.str();
            int fontface = FONT_HERSHEY_SCRIPT_SIMPLEX;
            double fontscale = 1.0;
            int baseline;
            Size textsize = getTextSize(text, fontface, fontscale, 2,
                                        &baseline);
            putText(frame, text, Point(det->c[0]-textsize.width/2,
                                       det->c[1]+textsize.height/2),
                    fontface, fontscale, Scalar(0xff, 0x99, 0), 2);
            //pose_estimation
            apriltag_detection_info_t info;
            info.det = det;
            info.tagsize = 178;
            info.fx = astra.fx;
            info.fy = astra.fy;
            info.cx = astra.u0;
            info.cy = astra.v0;
            apriltag_pose_t pose;

            double err = estimate_tag_pose(&info, &pose);
            cout<<err<<endl;
        }
        zarray_destroy(detections);
 ```
As the Error says, I suppose the function`matd_destroy` in `estimate_tag_pose` frees a invalid pointer.
> *** Error in `/home/xzy/Nursing_robot/camera/cmake-build-debug/Nursing_robot': free(): invalid pointer: 0x0000000000405b22 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f003f1d27e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f003f1db37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f003f1df53c]
/usr/local/lib/libapriltag.so.3(matd_destroy+0x45)[0x7f00451e2cac]
/usr/local/lib/libapriltag.so.3(estimate_tag_pose+0x83)[0x7f0044dbeee8]
/home/xzy/Nursing_robot/camera/cmake-build-debug/Nursing_robot[0x404c4d]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f003f17b830]
/home/xzy/Nursing_robot/camera/cmake-build-debug/Nursing_robot[0x403fd9]

ZhouYixuanRobtic, 2019-04-08:
Update, I try to comment the `matd_destroy` like below, and of course pose estimation runs well. But I don't think it's a good substitution solution, so waiting your generous help again. 
```c
double estimate_tag_pose(apriltag_detection_info_t* info, apriltag_pose_t* pose) {
    double err1, err2;
    apriltag_pose_t pose1, pose2;
    estimate_tag_pose_orthogonal_iteration(info, &err1, &pose1, &err2, &pose2, 50);
    if (err1 <= err2) {
        pose->R = pose1.R;
        pose->t = pose1.t;
        //matd_destroy(pose2.R);
        //matd_destroy(pose2.t);
        return err1;
    } else {
        pose->R = pose2.R;
        pose->t = pose2.t;
        //matd_destroy(pose1.R);
        //matd_destroy(pose1.t);
        return err2;
    }
}
```

ZhouYixuanRobtic, 2019-04-08:
update, I fix this problem when I looking at the #23 commit, while, I'm still confused about the unit, so waiting for your answers. Thanks again for your excellent again.

mkrogius, 2019-04-08:
Thanks for the post. It is in meters

closed 2019-04-08 by mkrogius

afoolboy, 2019-04-09:
I agree with him,I made the same mistake,I suppose the functionmatd_destroy in estimate_tag_pose frees a invalid pointer,too.
@ZhouYixuanRobtic @mkrogius 
If I use one image for attitude estimation, there is no problem, but if I use multiple images at the same time, there will be the following error
src/tcmalloc.cc:278] Attempt to free invalid pointer 0x3fdf1631a0f96086 
Aborted (core dumped)

ZhouYixuanRobtic, 2019-04-09:
check the #23 commit, this issue has been fixed, I've tried use multiple images, it works for me.@afoolboy

afoolboy, 2019-04-09:
谢谢你 thank you @ZhouYixuanRobtic

commit 37228502 (2019-04-08): Add comment describing units used. Fixes #24