AprilRobotics/apriltag #30: Problem About the Adaptive Threshold

Hxybit, 2019-04-19:
Hi,
I am glad to find this version shows the in-process pictures, it helps me a lot for learning this algorithm.
In the article ‘AprilTag 2: Efficient and robust fiducial detection’ fig.3 (b), the adaptive threshold result seemed normal. When I test the opencv demo, results are as follows. The right is the source picture, the left used another  algorithm to threshold, and put into detector. 
It looks so many noises and candidate quads when using threshold algorithm on the demo , so is this to save time or for precision?
Thanks!

[image]

mkrogius, 2019-04-19:
It looks like adaptive thresholding is performing particularly poorly here. Can you send me your source picture in full resolution?

On high resolution images where you do not care about detecting extremely small (apparent size of ~25px by 25px) tags then it is a good idea to downscale the source image before passing it to the detector.

Hxybit, 2019-04-20:
Thanks for your reply!
Effect of adaptive threshold should not much relationship with resolution. Your answer is enough for me to learn AprilTags, and the performance of the detector looks very good in test. Of course the video pipeline was in lower resolution.
[image]

mkrogius, 2019-04-27:
Hi, if the detector is now working as expected, would you please mark this issue as closed?

Thanks,
Max

closed 2019-04-28 by Hxybit