AprilRobotics/apriltag #35: How do I input an image for apriltag_demo?

JebBenson, 2019-05-10:
I've been able to build apriltag in a Linux VM and can run the apriltag_demo. But, when I run the program I don't have any options to select or any way to point to an image of an apriltag. All I get is an output that reads "Summary Hamm" with a bunch of zeros. Would appreciate any direction.

UberEclectic, 2019-05-10:
Hey,  Jeb,

I don't know the answer to your exact question, but have you tried the OpenCV example?  I found it very helpful, since it is interactive.

The OpenCV example shows you what the webcam sees and does live updates of the images by adding highlights to the detected markers.

Regards,
\Leon
--
Leon Shaner :: Dearborn, Michigan (iPad Pro)

pflashan, 2019-05-13:
If you were able to build apriltag_demo, calling it with -h will give you a simple usage output. Otherwise, just add an image file name to the commandline: ./apriltag_demo <image path>

The demo should support JPG, PPM, and PGM images, if I remember correctly.

JebBenson, 2019-05-13:
Thanks for both comments. I haven't installed OpenCV yet, but will try the opencv_demo as soon as I do. I added an image file to the command line. The tag was identified correctly, but I need to figure out what the rest of the results mean. I was hoping to get pose out of the demo, but I guess that's not the case.

pflashan, 2019-05-13:
You'll need your camera parameters to get pose. If you don't have a calibrated camera and know the camera intrinsics, OpenCV is probably your next step. You can access the homography matrix of a marker (where the homography is based on an ideal marker with corners at (1,1), etc.) without it, but to get pose from the library you need to feed it your camera parameters.

JebBenson, 2019-05-13:
Yes, after I made this comment I saw the section on pose estimation, so I will try according to those instructions. I am using a RealSense so I should be able to get the fx/fy/cx/cy.

closed 2019-05-13 by JebBenson