AprilRobotics/apriltag #37: [ROS2] Discussion for pthreads dependency on Windows

seanyen, 2019-05-22:
I am building `apriltag` from source code for ROS2 on Windows and I just realized it depends on `pthreads` libraries (most likely it means the one built from `vcpkg` after searching the git log). Without it, it will be failing at the following error messages:

```
Starting >>> apriltag
--- stderr: apriltag
--- output: apriltag
-- The C compiler identification is MSVC 19.16.27030.1
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
-- The CXX compiler identification is MSVC 19.16.27030.1
Please set them or make sure they are set and tested correctly in the CMake files:
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
PTHREAD_LIBRARIES
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
    linked by target "apriltag" in directory C:/colcon_ws/src/apriltag
-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done
---
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
Failed   <<< apriltag	[ Exited with code 1 ]
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
PTHREAD_LIBRARIES
    linked by target "apriltag" in directory C:/colcon_ws/src/apriltag

-- Configuring incomplete, errors occurred!
See also "C:/colcon_ws/build/apriltag/CMakeFiles/CMakeOutput.log".
---
```

And here are some ideas to make life little bit easier for Windows developers:

* Make this dependency to be called out in package.xml and maybe add `rosdep` key for this external thing.
* Or another example is to locally build `pthreads` as a local vendor library for this project, which can be found in https://github.com/ros2/rviz/tree/ros2/rviz_assimp_vendor.
* I haven't really looked into the code but maybe C++ code modernization could help get rid of the pthread dependency?

What do you think?

mkrogius, 2019-05-24:
Currently we only officially support Linux for AprilTags. Probably the best solution here is just to make sure that we call that out more explicitly in the documentation.

@wxmerkt Do you have any suggestions for making this clearer on the ROS side of things? Is there some way we can mark this as a linux-only package?

wxmerkt, 2019-05-24:
I’m afraid I’ve never built ros packages on windows. @seanyen do you have an example for the pthreads rosdep dependency? I couldn’t find a rosdep key. 
Given this is a pure C library I don’t think C++ threads modernisation is planned. 

(There are ways to blacklist packages but rare for people to do so - I haven’t received a build failure so I assume windows isn’t packaged via the build farm.)

mkrogius, 2019-05-28:
Also, this PR mentions one approach to building on windows: https://github.com/AprilRobotics/apriltag/pull/27

milijan, 2019-06-03:
@seanyen you could try to just stub out the pthread calls and set nthreads to 1 in the detector. If you don't need multithreads it should just work as far as I can tell.

matsuren, 2019-11-25:
Thanks for great project!
I deleted all pthread dependency and set nthread to 1. Everything works fine on Windows Visual Studio 2015. 
My source code is [here](https://github.com/matsuren/apriltag/tree/windows_nothread).

closed 2020-06-25 by mkrogius

1999kevin, 2020-11-10:
Hi, I'm trying to compile your code on Windows Visual Studio 2019. The compiling process is fine and the progress cannot decode jpg files. The fread command would not read the whole jpg file. Could you please help me figure why this happen?

212240447, 2022-05-24:
> Thanks for great project! I deleted all pthread dependency and set nthread to 1. Everything works fine on Windows Visual Studio 2015. My source code is [here](https://github.com/matsuren/apriltag/tree/windows_nothread).

Thank you for your analysis, let me solve the problem of msvc compiling on windows, thank you

christian-rauch, 2022-05-24:
There is PR https://github.com/AprilRobotics/apriltag/pull/95 by @ryancwicks to build without `pthreads` and implement Windows support. It needs to be rebased and cleaned up.