AprilRobotics/apriltag #44: streamline cmake build [non-standard and broken for non-Debian]

v4hn, 2019-07-04:
Hi there,

first of all thank you for providing the library. We use it a lot in our research. :+1: 

A few days ago you merged in a python wrapper as far as I can see and the patch [detects python in the most obscure way](https://github.com/AprilRobotics/apriltag/blob/master/CMakeLists.txt#L65) I have seen so far in my life as a distribution maintainer.
Please consider streamlining the detection process by simply using [`find_package(Python)` et. al.](https://cmake.org/cmake/help/latest/module/FindPython.html#module:FindPython).

The current process does not even mention NumPy, which you include in your sources though.
This accidentally works in Debian-based systems because they added a handy symlink
`/usr/include/numpy -> /usr/lib/python3.6/site-packages/numpy/core/include/numpy`, but this link is not defined through the numpy project and other Linux distributions do not have it.
So you really should explicitly request NumPy support.

On my system, I also see this other error, which is related to `lib` being called `lib64` on my system:
```
Errors     << apriltag:make ws/logs/apriltag/build.make.000.log
gcc: error: ws/build/apriltag/lib/libapriltag.so: No such file or directory
make[2]: *** [CMakeFiles/apriltag_python.dir/build.make:62: apriltag.cpython-36m-x86_64-linux-gnu.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/apriltag_python.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
I'm surprised this apparently does not appear on Debian-based systems, as they enforced this 64bit suffix quite strongly.

wxmerkt, 2019-07-04:
> I'm surprised this apparently does not appear on Debian-based systems, as they enforced this 64bit suffix quite strongly.

It's also the reason I have requested to revert the release to kinetic: It breaks builds on Xenial, while they work 'just fine' on Melodic.

zwn, 2019-07-15:
> The current process does not even mention NumPy, which you include in your sources though.
> This accidentally works in Debian-based systems because they added a handy symlink
> `/usr/include/numpy -> [...]

My build on ubuntu 18.04 just failed looking for numpy. Thanks to this hint I was able to fix it. My numpy is not even system installed, it is in venv. I believe the common way to build a python extension is to use setup.py which sidesteps the problem of python detection using cmake entirely and allows wheel generation, management with pip etc.

mkrogius, 2019-07-26:
1) Unfortunately find_package(Python) is not available at the most commonly available cmake version (3.10). I have made some improvements now so the numpy issue should be handled better, but I was not really able to test on my system. If you have ideas for how to better handle building this wrapper, but with the constraint that we support cmake 3.10, please let me know.

2) For this other error can you use make VERBOSE=1 or something like that? I'm not quite sure from the error messages here what exactly the issue is.

zwn, 2019-07-29:
It there is any interest, I can create some PR from my repository: https://github.com/zwn/apriltag. The changes are all aimed at being able to build and run the python extension on linux and windows with python 3. The instructions I have been sending my colleagues were:

```bash
git clone https://github.com/zwn/apriltag.git
cd apriltag
python3 -m venv env
source .env/bin/activate
pip install --upgrade pip
pip install cmake
mkdir build
cd build
cmake ..
cmake --build . --target pyinstall
```

For windows we have been adding `--config Release` to the last command. It needs cmake 3.14 for the numpy stuff but installing it with pip is convenient enough so it has not bee an issue so far.

Let me know if you want anything upstreamed.

mkrogius, 2019-08-01:
@zwn , Thanks for the offer, but we need to keep our CMakeLists version at 3.1 for now since that is what is available on ubuntu 18.04 right now.

closed 2020-06-25 by mkrogius