AprilRobotics/apriltag #45: Can't set black_border ?

HeYijia, 2019-07-05:
I want to use the 36h11 tags with 2 black borders, apriltag1 and apriltag2 can set this value. But I found that black_border is removed from apriltag3.
so, MY QUESTION IS: How can I use apriltag3 to detect 36h11 with 2 black_border.
thanks in advance.

mkrogius, 2019-07-26:
Hi,

We no longer allow directly changing the border width (it did not seem to be a useful option). There are two ways you could detect tags with a double wide black border now.

a) You can generate a new tag family with the desired layout using: https://github.com/AprilRobotics/apriltag-generation
This won't have the same pattern of bits though if you have tags already printed out that you need to detect.

b) You can hack the tag36h11.c file to make it detect tags with a double wide black border. Increase width_at_border and total_width by 2, and increase each of the bit_x/bit_y values by 1.

HeYijia, 2019-07-27:
Thank you for your suggestion.

closed 2019-07-27 by HeYijia

ishipachev, 2021-07-08:
Hi!
I've encountered the same problem when I was working with Kalibr.
Here is a patch to do changes mentioned above from border size 1 to border size 2.
[tag36h11_border1to2_patch.txt](https://github.com/AprilRobotics/apriltag/files/6785698/tag36h11_border1to2_patch.txt)

zpillio, 2023-01-11:
> Hi,
>
> We no longer allow directly changing the border width (it did not seem to be a useful option). There are two ways you could detect tags with a double wide black border now.
>
> a) You can generate [...]

I think kalibr is wildly enough used to say that double boarder feature was useful, so I would gently ask you to rethink it. :)

thx for the patch @ishipachev

giri-kum, 2024-01-11:
@zpillio  Could you mention where this code need to be placed?