AprilRobotics/apriltag #46: python warpper import location?

lvgeng, 2019-07-05:
Some issues.
I tried to import the python warpper from Ubentu 18.04 terminal command line.
It turns out that the only directory can do the 
> python3
> >>>import apriltag 
is the root folder
>
apriltag.c
apriltag.cpython-36m-x86_64-linux-gnu.so
apriltag_demo
apriltag_detect.docstring
apriltag_detect.docstring.h
apriltag.h
apriltag_math.h
apriltag.pc
apriltag.pc.in
apriltag_pose.c
apriltag_pose.h
apriltag_py_type.docstring
apriltag_py_type.docstring.h
apriltag_pywrap.c
apriltag_pywrap.o
apriltag_quad_thresh.c
CMakeCache.txt
CMakeFiles
cmake_install.cmake
CMakeLists.txt
common
example
install_manifest.txt
install.sh
lib
LICENSE.md
Makefile
package.xml
python_build_flags.py
README.md
tag16h5.c
tag16h5.h
tag25h9.c
tag25h9.h
tag36h11.c
tag36h11.h
tagCircle21h7.c
tagCircle21h7.h
tagCircle49h12.c
tagCircle49h12.h
tagCustom48h12.c
tagCustom48h12.h
tagStandard41h12.c
tagStandard41h12.h
tagStandard52h13.c
tagStandard52h13.h
>
The only way to successfully import apriltag is to make sure that the ./lib is in the same directory.
Is there any way to add the 
> libapriltag.so.3.1.0
into system lib path?

mkrogius, 2019-07-26:
Hi,

The library libapriltag.so.3.1.0 should be installed to your /lib folder after running the instruction for installing apriltag: https://github.com/AprilRobotics/apriltag#install

Does running these commands fix the problem?

joshvillbrandt, 2019-11-13:
I am having a similar problems. I have following the install instructions at https://github.com/AprilRobotics/apriltag#install

If I launch Python from the apriltag directory in which I installed from, then `from apriltag import apriltag` works fine.

If I try to do this import from my application folder ('/app'; I'm running this in Docker), then it cannot find apriltag. I got an warning during `make install` that said `cp: cannot create regular file '/root/.local/lib/python3.7/site-packages': No such file or directory` which is probably related. Everything except the final site-packages folder exists. I did `ln -s /usr/local /root/.local` to get around this. Re-ran `make install` and no warning this time.

Now I get a new error when trying to import:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: libapriltag.so.3: cannot open shared object file: No such file or directory
```

What should I do here?

Thanks!

cdb0y511, 2019-11-21:
> I am having a similar problems. I have following the install instructions at https://github.com/AprilRobotics/apriltag#install
> 
> If I launch Python from the apriltag directory in which I installed from, then `from apriltag import apriltag` works fine.
> 
> If I try to do this import from my application folder ('/app'; I'm running this in Docker), then it cannot find apriltag. I got an warning during `make install` that said `cp: cannot create regular file '/root/.local/lib/python3.7/site-packages': No such file or directory` which is probably related. Everything except the final site-packages folder exists. I did `ln -s /usr/local /root/.local` to get around this. Re-ran `make install` and no warning this time.
> 
> Now I get a new error when trying to import:
> 
> ```
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> ImportError: libapriltag.so.3: cannot open shared object file: No such file or directory
> ```
> 
> If I copy the /lib directory to my /app directory then the python import works. This isn't very clean and this step isn't in the install documentation, so I do not expect this to be the desired approach.
> 
> What should I do here? How is this supposed to work?
> 
> Thanks!

Hi, this is not a issue of the code. It's your environment.
just edit the ~/.bashrc add
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH 
then
source ~/.bashrc
and dont forget 
sudo ldconfig
It may solve your problem

joshvillbrandt, 2020-02-11:
Hey @cdb0y511,

That environment variable did the trick! I still needed the `ln -s /usr/local /root/.local` though.

Here is my working `dockerfile`:

```
FROM jjanzic/docker-python3-opencv:latest

ENV PYTHONUNBUFFERED 1
ENV PYTHONPATH "${PYTHONPATH}:/app"

# fix "ImportError: libapriltag.so.3: cannot open shared object file: No such file or directory"
# https://github.com/AprilRobotics/apriltag/issues/46
ENV LD_LIBRARY_PATH "/usr/local/lib:${LD_LIBRARY_PATH}"

WORKDIR /app
COPY . /app

RUN pip install --trusted-host pypi.python.org --upgrade pip \
    && pip install --trusted-host pypi.python.org -r requirements.txt \
    && git clone https://github.com/AprilRobotics/apriltag.git /home/apriltag \
    && cd /home/apriltag \
    # fix missing site-packages in /root/.local/lib/python3.7/
    && ln -s /usr/local /root/.local \
    && cmake . \
    && make install

CMD python app/app.py
```

Ultimately it would be great if I could simply do `pip install apriltag` and everything would just work. In the meantime, maybe some documentation updates would help other Python users get up and running.

closed 2020-06-25 by mkrogius