AprilRobotics/apriltag #49: fatal error: numpy/arrayobject.h: No such file or directory

lianghongzhuo, 2019-07-19:
while compiling the project, I encounter the following errors:
```
.../src/apriltag/apriltag_pywrap.c:6:31: fatal error: numpy/arrayobject.h: No such file or directory
compilation terminated.
```
I'm using ubuntu 16.04 and ros kinetic

mickeyouyou, 2019-07-21:
sudo apt-get install python-numpy

lianghongzhuo, 2019-07-21:
Thank you for your answer, but this doesn't work. I have python-numpy installed, but the code failed to find it.

EpsAvlc, 2019-07-26:
> Thank you for your answer, but this doesn't work. I have python-numpy installed, but the code failed to find it.

Have you solved the problem ? I met the error too and don't know how to do with it.

lianghongzhuo, 2019-07-26:
Check out to the version that did not add python wrapper of apriltag should work.
But this is only a workaround.

mkrogius, 2019-07-26:
Should be fixed now.

closed 2019-07-26 by mkrogius

thachdo, 2021-08-24:
The following command worked for me
```
sudo apt-get install --reinstall python-numpy
```