AprilRobotics/apriltag #50: note: use option -std=c99 or -std=gnu99 to compile your code

jimmccartney, 2019-07-19:
Trying to compile on Red Hat RHEL 7.  I've installed cmake3 which creates a makefile.  This error on make install.  Any ideas?
[make_errors.txt](https://github.com/AprilRobotics/apriltag/files/3411482/make_errors.txt)

mickeyouyou, 2019-07-21:
same cmake build error in ubuntu 18.04 x86.

jimmccartney, 2019-07-21:
I've made some more progress but still not complete.  I've changed 3 files:
`CMakeFiles/apriltag_demo.dir/flags.make:5:C_FLAGS = -O2 -DNDEBUG **-std=c99**  
CMakeFiles/apriltag_python.dir/build.make:68:	/bin/cc **-std=c99** -Wno-unused-result -Wsign-compare -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DNDEBUG -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -I/opt/rh/rh-python36/root/usr/include -g -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -D_GNU_SOURCE -fPIC -fwrapv -fPIC -I/opt/rh/rh-python36/root/usr/include/python3.6m -Wno-strict-prototypes -I/root/apriltag -c -o apriltag_pywrap.o /root/apriltag/apriltag_pywrap.c
CMakeFiles/apriltag.dir/flags.make:5:C_FLAGS = -O2 -DNDEBUG -fPIC **-std=c99** 
`
I also installed rh-python36-numpy package, as I am using software collections for python 3.6.
This got me all the way to the python
There is an error because the python build looks for the lib directory (I have lib64), so I soft-linked that.

Here is my last error:
`# make install
[ 83%] Built target apriltag
[ 89%] Built target apriltag_demo
[ 91%] Generating apriltag.cpython-36m-x86_64-linux-gnu.so
/bin/ld: cannot find -lpython3.6m
collect2: error: ld returned 1 exit status
make[2]: *** [apriltag.cpython-36m-x86_64-linux-gnu.so] Error 1
make[1]: *** [CMakeFiles/apriltag_python.dir/all] Error 2
make: *** [all] Error 2
`

Thanks!

mickeyouyou, 2019-07-22:
this error maybe you can try add command `link_directories(dir-to-libpython3.6.so)`

mickeyouyou, 2019-07-22:
I do some changes about default CMakeList.txt.  but stoped at

jimmccartney, 2019-07-22:
I got it compiled!  I exported LIBRARY_PATH=/opt/rh/rh-python36/root/usr/lib64, which is my $LD_LIBRARY_PATH.  Maybe it's working now.

`# make install
[ 83%] Built target apriltag
[ 89%] Built target apriltag_demo
[ 91%] Generating apriltag.cpython-36m-x86_64-linux-gnu.so
[100%] Built target apriltag_python
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib64/libapriltag.so.3.1.0
-- Installing: /usr/local/lib64/libapriltag.so.3
-- Installing: /usr/local/lib64/libapriltag.so
-- Installing: /usr/local/include/apriltag/apriltag.h
-- Installing: /usr/local/include/apriltag/apriltag_math.h
-- Installing: /usr/local/include/apriltag/apriltag_pose.h
-- Installing: /usr/local/include/apriltag/common/doubles.h
-- Installing: /usr/local/include/apriltag/common/doubles_floats_impl.h
-- Installing: /usr/local/include/apriltag/common/floats.h
-- Installing: /usr/local/include/apriltag/common/g2d.h
-- Installing: /usr/local/include/apriltag/common/getopt.h
-- Installing: /usr/local/include/apriltag/common/homography.h
-- Installing: /usr/local/include/apriltag/common/image_types.h
-- Installing: /usr/local/include/apriltag/common/image_u8.h
-- Installing: /usr/local/include/apriltag/common/image_u8x3.h
-- Installing: /usr/local/include/apriltag/common/image_u8x4.h
-- Installing: /usr/local/include/apriltag/common/matd.h
-- Installing: /usr/local/include/apriltag/common/math_util.h
-- Installing: /usr/local/include/apriltag/common/pam.h
-- Installing: /usr/local/include/apriltag/common/pjpeg.h
-- Installing: /usr/local/include/apriltag/common/pnm.h
-- Installing: /usr/local/include/apriltag/common/postscript_utils.h
-- Installing: /usr/local/include/apriltag/common/string_util.h
-- Installing: /usr/local/include/apriltag/common/svd22.h
-- Installing: /usr/local/include/apriltag/common/time_util.h
-- Installing: /usr/local/include/apriltag/common/timeprofile.h
-- Installing: /usr/local/include/apriltag/common/unionfind.h
-- Installing: /usr/local/include/apriltag/common/workerpool.h
-- Installing: /usr/local/include/apriltag/common/zarray.h
-- Installing: /usr/local/include/apriltag/common/zhash.h
-- Installing: /usr/local/include/apriltag/common/zmaxheap.h
-- Installing: /usr/local/include/apriltag/tag16h5.h
-- Installing: /usr/local/include/apriltag/tag25h9.h
-- Installing: /usr/local/include/apriltag/tag36h11.h
-- Installing: /usr/local/include/apriltag/tagCircle21h7.h
-- Installing: /usr/local/include/apriltag/tagCircle49h12.h
-- Installing: /usr/local/include/apriltag/tagCustom48h12.h
-- Installing: /usr/local/include/apriltag/tagStandard41h12.h
-- Installing: /usr/local/include/apriltag/tagStandard52h13.h
-- Installing: /usr/local/share/apriltag/cmake/apriltagConfig.cmake
-- Installing: /usr/local/share/apriltag/cmake/apriltagConfig-release.cmake
-- Installing: /usr/local/lib/pkgconfig/apriltag.pc
cp: cannot create regular file ‘/root/.local/lib/python3.6/site-packages’: No such file or directory
-- Installing: /usr/local/bin/apriltag_demo
-- Set runtime path of "/usr/local/bin/apriltag_demo" to ""
`

mickeyouyou, 2019-07-23:
congratulations @jimmccartney

mickeyouyou, 2019-07-23:
I think this is a python link problem, comment python warp for now, and build successfully.

If you are intrested about diff, here is :https://gist.github.com/mickeyouyou/93da4086d2c2d5ace02555c90e89130b

jimmccartney, 2019-07-23:
Okay, got it all working.  Here are my notes that I think are complete:
[Apriltag on RHEL7](https://github.com/jimmccartney/notes/blob/master/apriltag_rhel7.md)

mkrogius, 2019-07-26:
Sorry I didn't see this earlier. I've made some improvements to how the python wrapper works so it hopefully will now work without needing to set LIBRARY_PATH, but not totally sure.

closed 2019-07-26 by mkrogius

LamyZee, 2019-08-14:
Add 
`SET(CMAKE_C_FLAGS "-std=c99")
`
at CmakeLists.txt on line 3.