AprilRobotics/apriltag #52: AprilTags with IR/greyscale images?

a-price, 2019-07-27:
Would you expect to be able to detect tags in an image from an IR camera like the one on the Kinect2? I gave it a try but was unable to get any detections. (See image below for an example.)

[image]

atuleu, 2019-08-06:
This project  @formicidae-tracker/artemis  uses high resolution IR camera and apriltag3 to detect ants. There is no issue to use monochromatic images.

Maybe play with the quad detection algorithm parameter. First think to check is how many quads are detected in the image using the nquad field from apriltag_detector_t struct

Affie, 2019-09-06:
I quickly ran the detector on you image and it picked up 4 tags:
[image]

closed 2020-06-25 by mkrogius