AprilRobotics/apriltag #54: can't detect custom generated small standard tags

nzjrs, 2019-08-22:
I can't get apriltags3 (HEAD) to detect a newly created custom `java -cp april.jar april.tag.TagFamilyGenerator standard_5 5` border-less tags. That is despite the otherwise excellent improvements in performance that apriltags3 has over the previous version with small tags.

I put a complete test case and example data here https://gist.github.com/nzjrs/e951294289ac22320d7188572e804fb8

mkrogius, 2019-09-26:
Sorry for the slow response, it looks like the tags didn't render correctly. There should be one white square in the middle of the tags. I'm looking into the rendering code now

mkrogius, 2019-09-26:
Fixed, thanks for the bug report! https://github.com/AprilRobotics/apriltag-generation/commit/cea8bb1d38cd921ae621e998294cd51c8b44016e

nzjrs, 2019-10-02:
@mkrogius 

AFAICT, this commit does not fix the bug

The only change from the gist is I re-generated the C code with 

apriltag-generation version cea8bb1d38cd921ae621e998294cd51c8b44016e

The tag generator now generates 28 codes instead of 14. I have attached the generated `c`, `h`, and `java` files, along with the test case and tag mosaic.

can you reproduce?

[tagStandard16h5.c.txt](https://github.com/AprilRobotics/apriltag/files/3680830/tagStandard16h5.c.txt)
[tagStandard16h5.h.txt](https://github.com/AprilRobotics/apriltag/files/3680831/tagStandard16h5.h.txt)
[TagStandard16h5.java.txt](https://github.com/AprilRobotics/apriltag/files/3680833/TagStandard16h5.java.txt)
[image]

mkrogius, 2019-10-04:
Fixed. There was an issue with the detector where it was trying to fit a lighting model to the single pixel of white border in the middle of these tags.

closed 2019-10-04 by mkrogius