AprilRobotics/apriltag #55: memory leak using estimate_tag_pose

sindemcos, 2019-08-29:
I got a memory leak using estimate_tag_pose() and i am wondering if i forgot to free something. 
What I tried after receiving the pose:
 ```
matd_destroy(pose.R); 
matd_destroy(pose.t);
```

The valgrind output was:

```
7,992 bytes in 999 blocks are definitely lost in loss record 13,538 of 13,605
at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
by 0x92B3731: matd_svd_tall (in /usr/local/lib/libapriltag.so.3.1.0)
by 0x92B4781: matd_svd_flags (in /usr/local/lib/libapriltag.so.3.1.0)
by 0x92B47AE: matd_svd (in /usr/local/lib/libapriltag.so.3.1.0)
by 0x929DF6A: orthogonal_iteration (in /usr/local/lib/libapriltag.so.3.1.0)
by 0x92A0030: estimate_tag_pose_orthogonal_iteration (in /usr/local/lib/libapriltag.so.3.1.0)
by 0x92A0134: estimate_tag_pose (in /usr/local/lib/libapriltag.so.3.1.0)
```
Is it my bad or could there be a memory leak in the library?

mkrogius, 2019-09-27:
That is the correct way to free it. I've fixed the bug with the memory leak.

closed 2019-09-27 by mkrogius

Utsav13, 2019-11-25:
I have the same issue in my python code, so how can I free memory in python code.
I am using apriltags3 library.

mkrogius, 2019-11-25:
I believe the issue should be fixed as of Sept 27. If you are still experiencing this issue with the most recent version of the code from the repo, then please post a stack trace/valgrind output for your issue.

Utsav13, 2019-11-26:
okay, I got it, I have downloaded an older version of this code, after updating its work fine.
Thank you.

ShaunCurrier, 2022-05-13:
Is the library user required to call `matd_destroy()` on the pose members (as shown in the first post) after receiving and using the pose?