AprilRobotics/apriltag #56: Cannot detect tag36h11 from Asus Xtion color stream feed

charkops, 2019-09-06:
Hello, i am trying to detect a tag36h11 printed on a piece of paper using an Asus Xtion pro live camera. 
Although i can easily detect the printed tag when i capture frames from my webcam, i cannot detect it when using the Xtion.

I specifically start the colour stream from the Xtion using openni2, capture a rgb frame, convert to grayscale and then run the apriltag_detector, but to no avail. 

If i redirect the feed from my webcam to the programm instead, the detection works as expected. 

Anyone ever had any issues similar to this one ? 

I will attach code and a sample image, in which i cannot detect anything

Thanks

[image]

```
#include <iostream>

#include "opencv2/opencv.hpp"
#include "opencv2/core/core.hpp"
#include "opencv2/imgcodecs.hpp"
#include "opencv2/highgui/highgui.hpp"
#include <OpenNI.h>

extern "C" {
#include "apriltag.h"
#include "tag36h11.h"
}

// Helper functions
void detectAprilTag(const cv::Mat &);

int main(int argc, char** argv){
 	const openni::SensorType sensorType = openni::SENSOR_COLOR;
 	const char windowName[] = "AprilTagDetection";

 	int srcMatType;
 	if(sensorType == openni::SENSOR_COLOR){
 		srcMatType = CV_8UC3;
 	} else {
 		srcMatType = CV_16U;
 	}

 	// Initialize camera, load drivers
 	openni::Status status;
 	status = openni::OpenNI::initialize();
 	if(status != openni::STATUS_OK){
 		printf("Failed to initialize OpenNI:\n%s\n",openni::OpenNI::getExtendedError());
 		return EXIT_FAILURE;
 	}

 	// Connect to any available OpenNI-compliant camera
 	openni::Device device;
 	status = device.open(openni::ANY_DEVICE);
 	if (status != openni::STATUS_OK){
 		printf("Failed to open device: \n%s\n", openni::OpenNI::getExtendedError());
 		openni::OpenNI::shutdown();
 		return EXIT_FAILURE;
 	}

 	// Ensure that device has the appropriate type of sensor
 	const openni::SensorInfo *sensorInfo = device.getSensorInfo(sensorType);
 	if(sensorInfo == NULL){
 		printf("Failed to find sensor of appropriate type\n");
 		device.close();
 		openni::OpenNI::shutdown();
 		return EXIT_FAILURE;
 	}

 	// Create stream, don't start it yet
 	openni::VideoStream stream;
 	status = stream.create(device, sensorType);
 	if(status != openni::STATUS_OK){
 		printf("Failed to create stream: \n%s\n", 
 			openni::OpenNI::getExtendedError());
 		device.close();
 		openni::OpenNI::shutdown();
 		return EXIT_FAILURE;
 	}
	
	{
	 	// Select video mode with highest resolution
	 	const openni::Array<openni::VideoMode> *videoModes = &sensorInfo->getSupportedVideoModes();
	 	int maxResolutionX = -1;
	 	int maxResolutionIndex = 0;
	 	for(int i = 0; i < videoModes->getSize(); i++){
	 		int resolutionX = (*videoModes)[i].getResolutionX();
	 		if(resolutionX > maxResolutionX){
	 			maxResolutionX = resolutionX;
	 			maxResolutionIndex = i;
	 		}
	 	}
	 	stream.setVideoMode((*videoModes)[maxResolutionIndex]);
 	}
	

	openni::VideoMode vm = stream.getVideoMode();
	int cols, rows;
	cols = vm.getResolutionX();
	rows = vm.getResolutionY();

 	// Start stream
 	status = stream.start();
 	if (status != openni::STATUS_OK){
 		printf("Failed to start stream: \n%s\n",
 			openni::OpenNI::getExtendedError());
 		stream.destroy();
 		device.close();
 		openni::OpenNI::shutdown();
 		return EXIT_FAILURE;
 	}

 	// OpenNI-frame, OpenCV-matrix and an openCV window
 	openni::VideoFrameRef frame;
 	cv::Mat dstMat;
 	cv::namedWindow(windowName);

	// TODO (charis): Delete this as soon we debug this
	// Initialize camera
	/*
	cv::VideoCapture cap(0);
	if (!cap.isOpened()) {
	    std::cerr << "Couldn't open video capture device" << '\n';
	    return -1;
	}
	*/

 	// Capture and display frames until any key is pressed
 	while(cv::waitKey(1) == -1){
 		status = stream.readFrame(&frame);
 		if(frame.isValid()){
		    cv::Mat srcMat (
				    frame.getHeight(), frame.getWidth(), srcMatType, 
				    (void *)frame.getData(), frame.getStrideInBytes());
		    if(sensorType == openni::SENSOR_COLOR){
			cv::cvtColor(srcMat, dstMat, cv::COLOR_RGB2BGR);
		    } else {
			srcMat.convertTo(dstMat, CV_8UC1);
		    }
		    // openni::RGB888Pixel* cData = (openni::RGB888Pixel *)frame.getData();
		    // cv::Mat colorImage(rows, cols, CV_8UC3, cData);
			

		    // dstMat = cv::imread("/home/xbaremenos/programms/apriltag/example/gray_frame.jpeg");
		    cv::imshow(windowName, dstMat);

		    // Detect april Tag in dstMat
		    // cap >> dstMat;
		    // std::cerr << "dstMat_type = " << dstMat.type() << '\n';
		    detectAprilTag(dstMat);
 		}
 	}

 	cv::destroyWindow(windowName);

 	stream.stop();
 	stream.destroy();
 	device.close();
 	openni::OpenNI::shutdown();
 	return 0;

 }

void detectAprilTag(const cv::Mat &frame){
    apriltag_detector_t *td = apriltag_detector_create();
    apriltag_family_t *tf = tag36h11_create();
    apriltag_detector_add_family(td, tf);

    
    td->quad_decimate = 2.0;
    td->quad_sigma = 0.5;
    td->refine_edges = 1;
    td->decode_sharpening = 0.25;
    td->nthreads = 4;
    // td->debug = 1;

    // Detection
    // Make sure we have a grayscale img
    cv::Mat gray;
    if(frame.type() == CV_8UC1) gray = frame;
    else cv::cvtColor(frame, gray, cv::COLOR_BGR2GRAY);
	    
    // cv::cvtColor(frame, gray, cv::COLOR_RGB2GRAY);
    // gray = cv::imread("/home/xbaremenos/programms/apriltag/example/gray_frame.jpeg", 0);

    // cv::GaussianBlur(gray, gray, cv::Size(15, 15), 0);
    // std::cerr << "gray_type = " << gray.type() << '\n';
    cv::imshow("gray", gray);

    image_u8_t im = { 
		      .width = gray.cols,
		      .height = gray.rows,
		      .stride = gray.cols,
		      .buf = gray.data
    };

    zarray_t *detections = apriltag_detector_detect(td, &im);
        
    std::cout << zarray_size(detections) << " tags detected" << '\n';
    // std::cout << "td->quad_sigma = " << td->quad_sigma << '\n';
    
    /*
    std::cout << "td->quad_decimate = " << td->quad_decimate << '\n';
    std::cout << "td->quad_sigma = " << td->quad_sigma << '\n';
    std::cout << "td->refine_edges = " << td->refine_edges << '\n';
    std::cout << "td->decode_sharpening = " << td->decode_sharpening << '\n';
    std::cout << "td->nthreads = " << td->nthreads << '\n';
    std::cout << "td->debug = " << td->debug << '\n';
    */

    // Cleanup
    apriltag_detections_destroy(detections);
    apriltag_detector_destroy(td);
    tag36h11_destroy(tf);
    
}
```

Affie, 2019-09-06:
your image is flipped
[image]

charkops, 2019-09-06:
Oops
Well, that is embarassing :)

Thank you for noticing that!

closed 2019-09-06 by charkops