AprilRobotics/apriltag #57: can't detect custom generated small standard tags

nzjrs, 2019-10-02:
I'm not sure if you get notifications on closed bugs but I couldn't re-open #54, so filing this. Please close if that is preferred.

Thanks a lot.

closed 2019-10-02 by mkrogius

nzjrs, 2019-10-06:
Thanks a lot. I can confirm this fixes the bug and doesn't introduce any other regressions for detecting other dag dictionaries

(our CI for our small tags is now passing)

mkrogius, 2019-10-06:
Great! If you don't mind, can I ask what application you have in mind for these small tags?

nzjrs, 2019-10-22:
@mkrogius sorry for the delay. We have a cloud/browser based image processing product for biologists and we offer apriltags as a tracking option. I actually recently tweeted about this here https://twitter.com/loopbiogmbh/status/1186009440404639744 (there are tags on the bugs on the left)

Incidentally, we always reference and cite the original publication in algorithms we offer to users, and encourage them to do the same. Do you have a citation / paper for apriltags3 yet? The most recent one I have is the 2016 one.

mkrogius, 2019-10-24:
Very cool! One thing to be aware of with these tiny tags is that since they have a very small number of bits, you probably will run into some amount of false positives. I would definitely recommend increasing the number of bits in the tag if possible (more bits means less detection distance).

There is a citation and a paper which will be published at IROS in two weeks. The paper can be found at https://april.eecs.umich.edu/papers/details.php?name=krogius2019iros
It can be cited as:
@inproceedings{krogius2019iros,
    TITLE      = {Flexible Layouts for Fiducial Tags},
    AUTHOR     = {Maximilian Krogius and Acshi Haggenmiller and Edwin Olson},
    BOOKTITLE  = {Proceedings of the {IEEE/RSJ} International Conference on Intelligent
                 Robots and Systems {(IROS)}},
    YEAR       = {2019},
    KEYWORDS   = {Localization},
}