AprilRobotics/apriltag #59: Error, more than one new minima found

allsey87, 2019-10-10:
Regarding the fprintf in [apriltag_pose.c:435](https://github.com/AprilRobotics/apriltag/blob/3.1.1/apriltag_pose.c#L435) that prints "Error, more than one new minima found" if the previous pose estimate was not good, how is a user of your library supposed to react to this message.

Is this bug or is it debugging output that can be ignored? In the latter case, perhaps this debugging output can be toggled on and off using a member of the apriltag_detection_info_t struct?

hwscut, 2019-10-23:
When seeing more than one tag to estimate tag_pose, I also get such error. 
waitting for solution.

zftan0709, 2020-04-15:
`Tag ID: 10
Normalized： 
-1.76278 -0.31017
Translation: 
4.97972  -0.959973 -2.89562
Rotation: 
-0.928616 0.175963 -0.326666
0.13727 0.980855 0.138131
-0.344717 -0.0834289 0.934992

Error, more than one new minimum found.

Tag ID: 10
Normalized： 
-1.76919 -0.31115
Translation: 
-5.06095  0.973719 2.93369
Rotation: 
-0.156745 -0.142866 0.977251
0.0681974 -0.988686 -0.133599
0.985282 0.0457049 0.164715
`
I'm having the same problem and the Tz changes to negative from time to time.

C-monC, 2020-09-18:
Hi,

I get this error with only one tag in the image. Did anyone figure out the reason for this error?

wxmerkt, 2022-03-10:
Can you provide a test image where this happens?

mpillet, 2024-01-31:
Can we reopen this issue please ? I'm facing the same problem with 'Error, more than one new minima found' leading to a wrong reconstruction

christian-rauch, 2024-01-31:
> Can we reopen this issue please ? 

I think this issue was never closed. So no worries there :-)

> I'm facing the same problem with 'Error, more than one new minima found' leading to a wrong reconstruction. Here is an example where the detection of the tag works but with this error of more than one new minima found.

Is this reproducible with the `apriltag_demo`? Can you provide a jpeg image and potential command-line arguments that cause the issue with the `apriltag_demo`?

ggb367, 2024-02-15:
Hi @christian-rauch, we're also running into this multiple minima error with this image when estimating the pose of each of the tags: 
[image]
Usually one minima is correct while the other is quite incorrect.
We tried to see if we could reproduce the error with aprilrtag_demo with:
`./apriltag_demo -f tagStandard52h13 -i 10 ./image_raw.jpg`
But we don't see a way to estimate the pose with that script.
Let me know if I can provide any additional details to figure this issue out.
Thanks!

christian-rauch, 2024-02-15:
> We tried to see if we could reproduce the error with aprilrtag_demo with: `./apriltag_demo -f tagStandard52h13 -i 10 ./image_raw.jpg` But we don't see a way to estimate the pose with that script.

Thanks for sharing the example image. The `apriltag_demo` executable only does the tag detection without the pose estimation.

> Let me know if I can provide any additional details to figure this issue out. Thanks!

Can you also provide the camera intrinsics used for the pose estimation? Is the code you use publicly available?

ggb367, 2024-02-15:
> Is the code you use publicly available?    

The repository is not public, but I can share code snippets as needed

> Can you also provide the camera intrinsics used for the pose estimation?    

See below:
```YAML
image_width: 640
image_height: 480
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [661.97625,   0.     , 313.69227,
           0.     , 661.02677, 257.01043,
           0.     ,   0.     ,   1.     ]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [0.054322, -0.183210, 0.001954, -0.002335, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1., 0., 0.,
         0., 1., 0.,
         0., 0., 1.]
projection_matrix:
  rows: 3
  cols: 4
  data: [662.69727,   0.     , 312.00752,   0.     ,
           0.     , 662.65527, 257.08592,   0.     ,
           0.     ,   0.     ,   1.     ,   0.     ]
```

christian-rauch, 2024-02-16:
I tried reproducing this by calling `estimate_tag_pose` on the detections in the `apriltag_demo.c` on the current `master` (51466ec524d9a5d7b759f42663e5da9810ff5261):
```cpp
for (int i = 0; i < zarray_size(detections); i++) {
    apriltag_detection_t *det;
    zarray_get(detections, i, &det);

    if (!quiet)
        printf("detection %3d: id (%2dx%2d)-%-4d, hamming %d, margin %8.3f\n",
                i, det->family->nbits, det->family->h, det->id, det->hamming, det->decision_margin);

    hamm_hist[det->hamming]++;
    total_hamm_hist[det->hamming]++;

    // pose estimation
    apriltag_detection_info_t info;
    info.det = det;
    info.tagsize = 0.05;
    info.fx = 661.97625;
    info.fy = 661.02677;
    info.cx = 313.69227;
    info.cy = 257.01043;
    apriltag_pose_t pose;
    estimate_tag_pose(&info, &pose);
    printf("  >> posistion: %f %f %f\n", pose.t->data[0], pose.t->data[1], pose.t->data[2]);
}
```

This gives me 8 detections with a pose (just printing position here):
```
detection   0: id (52x13)-4   , hamming 0, margin  141.337
  >> posistion: 0.043182 -0.045479 1.442508
detection   1: id (52x13)-6   , hamming 0, margin  151.679
  >> posistion: -0.064532 -0.246879 1.471909
detection   2: id (52x13)-7   , hamming 0, margin  141.630
  >> posistion: -0.066965 -0.143569 1.462468
detection   3: id (52x13)-8   , hamming 0, margin  158.549
  >> posistion: 0.145398 -0.044111 1.447844
detection   4: id (52x13)-10  , hamming 0, margin  147.847
  >> posistion: 0.036983 -0.148168 1.462753
detection   5: id (52x13)-11  , hamming 0, margin  148.997
  >> posistion: 0.144882 -0.143754 1.466733
detection   6: id (52x13)-12  , hamming 0, margin  157.467
  >> posistion: 0.137947 -0.247066 1.487945
detection   7: id (52x13)-14  , hamming 0, margin  145.122
  >> posistion: -0.060477 -0.038616 1.433352
detection   8: id (52x13)-16  , hamming 0, margin  153.694
  >> posistion: 0.036604 -0.253608 1.481749
```

I am using the camera parameters from the `camera_matrix` above and guessed the tag size to be 5cm. I built in debug mode so that `debug_print` will print to `stderr` (otherwise this will be silent), but I don't see the `"Error, more than one new minimum found.` message on the screen.

@ggb367 Can you add this to the apriltag_demo as well, update the tag size and camera parameters if needed and also tell me the remaining missing parameters to reproduce this issue?

christian-rauch, 2024-05-09:
Since there is no feedback on this, I assume that the issue has been resolved on your side. If it still persists, please reopen with more information to reproduce the issue.

closed 2024-05-09 by christian-rauch