AprilRobotics/apriltag #61: TagStandard52h13 and TagCustom48h12 is not getting detected

optimus-droid, 2019-11-05:
I have no issue with fewer number of tag id's like Tag36h10, Tag36h11 but when I tried with TagStandard52h13 with help of JNI function but my application got crashed. Please help, This is the first time I am using Apriltag. I have followed all the steps as guided in the wiki. There is something wrong in my process while using TagStandard52h13.

I have followed the steps like below.

image_u8_t* im = image_u8_create_from_pnm("test.png");
apriltag_detector_t *td = apriltag_detector_create();
apriltag_family_t *tf = tagStandard52h13_create();
apriltag_detector_add_family(td, tf);
zarray_t *detections = apriltag_detector_detect(td, im);

for (int i = 0; i < zarray_size(detections); i++) {
    apriltag_detection_t *det;
    zarray_get(detections, i, &det);

    // Do stuff with detections here.
}
// Cleanup.
tagStandard52h13_destroy(tf);
apriltag_detector_destroy(td);

But no tags are being detected.

iTronTik, 2020-01-09:
Hello, 

Are you able to detect it? I am getting an error saying reduce the max hemming distance.

andnovar, 2020-05-15:
Once you reduce the max hamming distance is still breaking. I have been looking into the code and seems like a problem at initialization. family->impl is not referenced when passing to apriltag_detector_add_family_bits in apriltag.c . I guess is something with the tagCustom48h12_create function. However, it runs normally if executed directly from a C project but not as a dll called from an external program.

easedroid, 2020-05-17:
> Once you reduce the max hamming distance is still breaking. I have been looking into the code and seems like a problem at initialization. family->impl is not referenced when passing to [...]

Are you running this on Android? If yes then this tag family won't work directly on android due to memory issue. I have tweaked it little bit and it's working on android. I will soon publish it to github

mkrogius, 2020-12-17:
Yes, the larger tag families can cause memory issues. You can try setting the max hamming distance to zero. Another option is to simply reduce the number of tags in that family (by editing the code, there's no more user-friendly way to reduce the number of tags yet)

closed 2020-12-17 by mkrogius