AprilRobotics/apriltag #62: How to build for python2?

MikoyChinese, 2019-11-11:
Hi,

I git clone this repository and to build it. I can import the apriltag in python3 but can't import in python2. I check the CMakeList.txt file, and found that `# TODO deal with both python2/3`.

So I want to know how can I build it for python2?

Thanks,
Mikoy Chinese

MikoyChinese, 2019-11-12:
It works for me to change the `CMakeList.txt` as following:

```
# Python wrapper
SET(Python_ADDITIONAL_VERSIONS 2)
find_package(PythonLibs 2.7 REQUIRED)
execute_process(COMMAND which python2 OUTPUT_QUIET RESULT_VARIABLE Python2_NOT_FOUND)
execute_process(COMMAND python2 -c "import numpy" RESULT_VARIABLE Numpy_NOT_FOUND)
if (NOT Python2_NOT_FOUND AND NOT Numpy_NOT_FOUND AND PYTHONLIBS_FOUND)
# TODO deal with both python2/3
execute_process(COMMAND python2 ${CMAKE_SOURCE_DIR}/python_build_flags.py OUTPUT_VARIABLE PY_OUT)
set(PY_VARS CFLAGS LDFLAGS LINKER EXT_SUFFIX)
cmake_parse_arguments(PY "" "${PY_VARS}" "" ${PY_OUT})
separate_arguments(PY_CFLAGS)
separate_arguments(PY_LDFLAGS)
set(LIB_APRILTAG_PATH ${PROJECT_BINARY_DIR}/lib)

foreach(X detect py_type)
add_custom_command(OUTPUT ${PROJECT_BINARY_DIR}/apriltag_${X}.docstring.h
  ┊ COMMAND < ${CMAKE_SOURCE_DIR}/apriltag_${X}.docstring sed 's/\"/\\\\\"/g\; s/^/\"/\; s/$$/\\\\n\"/\;'
  ┊ WORKING_DIRECTORY ${PROJECT_BINARY_DIR})
endforeach()
                                                                                                         
add_custom_command(OUTPUT apriltag_pywrap.o
  ┊ COMMAND ${CMAKE_C_COMPILER} ${CMAKE_C_FLAGS} ${PY_CFLAGS} -I${PROJECT_BINARY_DIR} -c -o apriltag_pywr
  ┊ DEPENDS ${CMAKE_SOURCE_DIR}/apriltag_pywrap.c ${PROJECT_BINARY_DIR}/apriltag_detect.docstring.h ${PRO
add_custom_command(OUTPUT apriltag${PY_EXT_SUFFIX}
  ┊ COMMAND ${PY_LINKER} ${PY_LDFLAGS} -Wl,-rpath=lib apriltag_pywrap.o ${LIB_APRILTAG_PATH}/libapriltag.
  ┊ DEPENDS ${PROJECT_NAME} apriltag_pywrap.o)
add_custom_target(apriltag_python ALL 
  ┊ DEPENDS apriltag${PY_EXT_SUFFIX})

execute_process(COMMAND python2 -m site --user-site OUTPUT_VARIABLE PY_DEST)
string(STRIP ${PY_DEST} PY_DEST)
install(CODE "execute_process(COMMAND cp ${PROJECT_BINARY_DIR}/apriltag${PY_EXT_SUFFIX} ${PY_DEST})")
endif (NOT Python2_NOT_FOUND AND NOT Numpy_NOT_FOUND AND PYTHONLIBS_FOUND)
```

jonazpiazu, 2019-11-13:
See https://github.com/AprilRobotics/apriltag/pull/64

MikoyChinese, 2019-11-14:
Thanks for your commit. You are a outstanding coder.

christian-rauch, 2022-06-03:
I am closing this because we [do not plan to support Python 2](https://github.com/AprilRobotics/apriltag/pull/64#issuecomment-756424560).

closed 2022-06-03 by christian-rauch