AprilRobotics/apriltag #63: AprilTag Detection and Pose Estimation

batuhanakcay, 2019-11-13:
We are new to AprilTags and struggling with detection and pose estimation of apriltags from images.

We have installed the code for AprilTags from https://github.com/AprilRobotics/apriltag on a Google Colab Notebook, then on Google Colab Notebook we utilize the AprilTag python wrapper and execute AprilTag detection from images. Our image inputs are photos of AprilTags printed on standard paper taken from smartphones and computer webcameras. We are not able to detect the AprilTags in our images accurately, it seems like the output position information we get using the detector.detect() function is inaccurate. However, when we run the AprilTags IOS app on our images, the IOS app accurately detects the tag and estimates pose. We are not sure what is the problem.

We have shared our python code below, attached our inputs and outputs. Could you help us out about how to use AprilTags functions properly and solve our problem with detection and pose estimation?

Python Code:

import cv2
import numpy as np
from apriltag import apriltag
from matplotlib import pyplot as plt

imagepath = 'april_cam.jpg'
ii = np.zeros((25,25))
image = plt.imread(imagepath, cv2.IMREAD_GRAYSCALE)
image = image[:,:,0].copy()
plt.imshow(image)

detector = apriltag("tagStandard41h12")
print(detector)
detections = detector.detect(image)
ii = image.copy()
width = 25
for i in range(len(detections)):
  center = detections[i]['center']
  bound = detections[i]['lb-rb-rt-lt']
  print(bound)
  for bb in bound:
    ii[int(bb[0])-width:int(bb[0])+width,int(bb[1])-width:int(bb[1])+width] = 1
  ii[int(center[0])-width:int(center[0])+width,int(center[1])-width:int(center[1])+width] = 1
plt.imshow(ii)

[image]
[image]
[image]

mkrogius, 2019-11-13:
Thank you for posting the code and images, that definitely helps me out.

It looks to me like the issue is with how you render the tag detections. Both bb and center are of the form (x,y), not (y,x). In other words your third to last line should be:
ii[int(bb[1])-width:int(bb[1])+width,int(bb[0])-width:int(bb[0])+width] = 1

batuhanakcay, 2019-11-13:
Hi Max,

Thank you for pointing this out. We have made the necessary changes to our
code and we got the desired results. AprilTag works very well and its
features are very impressive. Thanks for your help and working on such
a great project.

If we have any other questions, we will let you know by either posting on
github repo or sending an email.

Sincerely,
Batuhan Akcay
MDP Assistive Vision Team

mkrogius, 2019-11-25:
Great, thanks!

closed 2019-11-25 by mkrogius