AprilRobotics/apriltag #66: Query about 3D coordinate frame of detected Tag pose

nrasulnrasul, 2019-12-17:
Hi,
I would like to know the coordinate system of pose resulting from `estimate_tag_pose`. (x-right, y-down, z- goes inside the tag )?. 

Also what is the correct sequence(`XYZ` or `ZYX` or ...) to determine roll, pitch, yaw from rotation matrix of pose?.

Thank you.

matiasfrosini, 2020-05-08:
Hi, 
I have the same question. Could you find the answer ?

Thank you.

alddiaz, 2020-05-09:
Hi guys,
Q1: Yes, you are right. The coordinates of the tag are equal to the camera coordinates, i.e., x->right, y->down, z->inside the tag. You should be able to contrast this against the estimated 3-D tag orientation / rotation matrix.

Q2: The sequence of Euler angles doesn't matter as soon as you preserve it through all your calculations. You can use any one. A very typical sequence used in many engineering applications is XYZ, i.e., first rotation around z-axis (yaw), then rotation around y-axis (pitch), and finally around x-axis (roll).

artificial-cassiano, 2020-05-29:
did anyone get the poses using the python binding already?

alddiaz, 2020-05-29:
nope, but I rather used a MATLAB implementation I wrote for that purpose, perhaps it could be helpful....
https://github.com/alddiaz/MATLAB_AprilTag3

yuzhiyiliu, 2020-07-22:
> Hi guys,
> Q1: Yes, you are right. The coordinates of the tag are equal to the camera coordinates, i.e., x->right, y->down, z->inside the tag. You should be able to contrast this against the estimated [...]

Are you sure?

yuzhiyiliu, 2020-07-24:
I'm sorry, my tag got left hand coordinate system, maybe x->right, y->down, z->outside.
pose_R :
[[ 0.9999614  -0.00163934 -0.00863229]
 [-0.00865461 -0.01413587 -0.99986263]
 [-0.00151709 -0.99989874  0.01414951]]
pose_t :
[[0.00151295 0.14521985 0.23501338]]

mkrogius, 2020-08-02:
The pose refers to the position of the tag within the camera frame, not the position of the camera relative to the tag. Maybe this is why it appears left-handed?

yuzhiyiliu, 2020-08-03:
The position of my tag within the camera frame, I am sure.

mkrogius, 2020-12-17:
I have added information regarding the coordinate frame to the user guide: https://github.com/AprilRobotics/apriltag/wiki/AprilTag-User-Guide#coordinate-system

closed 2020-12-17 by mkrogius