AprilRobotics/apriltag #68: Detect tagCircle49h12 error

Happyan47, 2019-12-28:
When i want to detect tagCircle49h12 in opencv_demo,there has error:
apriltag.c: failed to allocate hamming decode table. Reduce max hamming size.

I only use "getopt_add_string(getopt, 'f', "family", "tagCircle49h12", "Tag family to use");" replace "getopt_add_string(getopt, 'f', "family", "tag36h11", "Tag family to use");" in thre opencv_demo.

How do you detect tagCircle49h12?

Happyan47, 2020-01-02:
It's worked out，thank you

iTronTik, 2020-01-09:
Hello, 
I am also facing similar issue for 52h13 tag. It says failed to allocate hamming decode table. Reduce max hamming size. Could you please let me know how did you solve it?

Happyan47, 2020-01-10:
> Hello,
> I am also facing similar issue for 52h13 tag. It says failed to allocate hamming decode table. Reduce max hamming size. Could you please let me know how did you solve it? This is not happening with lower tag families such as 36h11.

In opencv_demo.cpp 
apriltag_detector_add_family(detector, tag_family)  change to apriltag_detector_add_family_bits(detector, tag_family, 1), only detect tagCircle49h12.I think 52h13 should be true, you can try it.
when you want to detect tag32h11,you shoule use apriltag_detector_add_family(detector, tag_family)

iTronTik, 2020-01-22:
Hello Happyan,
Yes, it's working. Thanks.

closed 2020-12-17 by mkrogius