AprilRobotics/apriltag #69: Is goodness still represented anywhere?

jbendes, 2020-01-17:
Apriltag 2 had a concept of `goodness` defined as a measure of the quality of tag localization - average contrast of the pixels around the border of the tag. Does this concept still exist anywhere?

jbendes, 2020-01-17:
Additionally there was a `refine_pose` that enabled that computation if that helps shed any light

jbendes, 2020-01-17:
Is it essentially equivalent to the return of the following?
`double estimate_tag_pose(apriltag_detection_info_t* info, apriltag_pose_t* pose);`

mkrogius, 2020-06-16:
There is no longer an estimate of goodness. I found that it did not really provide too much information about whether detection was successful or whether the tag's pose was accurate. If you need super accurate tag localizations and want to throw out some detections, it is probably better to use some other heuristic.

closed 2020-12-17 by mkrogius