AprilRobotics/apriltag #75: Any tips to increase detection rate?

suraj2596, 2020-02-23:
Hi all,

Currently, I'm able to detect tags at 15fps on Pi 4 on 480x480 frames. Though 'image_raw' publishes up to 50fps, the '/tag_detection' rate doesn't change even though a lot of RAM is available. Tried overclocking but the detection rate fell down on contrary. And can't really reduce tag sizes anymore.

Any advice from your past experiences? 

Thanks!

suraj2596, 2020-02-23:
EDIT: In order to decimate, `tag_decimate` needs to be increased in value. Silly me, I was doing the opposite.

mkrogius, 2020-06-16:
Sorry for not responding sooner. Was your effort successful? What frame rate did you achieve?

suraj2596, 2020-06-16:
Hey! No problem.

Playing with the decimate tag gave up to 30fps. That's pretty decent I suppose.

mkrogius, 2020-06-17:
Yeah, that's not bad, but neither is it that good. What did you set the decimate factor to?

suraj2596, 2020-06-21:
Made it 1.5. But on a laptop, I used to get 30fps for default settings. What else can we expect from Pi's limited CPU.

closed 2021-05-07 by mkrogius