AprilRobotics/apriltag #78: Missing target Threads::Threads

stefanbuettner, 2020-03-13:
Hey there,

I was just about to use the apriltag library in a cmake project and encountered the following error message:
```
CMake Error at CMakeLists.txt:10 (add_library):
  Target "apriltag-test" links to target "Threads::Threads" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
```

Shouldn't the `apriltagConfig.cmake` handle finding the Threads library if it's a public dependency? A minimal example can be found [here](https://github.com/stefanbuettner/apriltag-cmake-configuration-failure-example).

Cheers,
Stefan

mkrogius, 2020-07-10:
Thanks for the example!

closed 2020-07-10 by mkrogius

christianrauch, 2020-08-04:
Note: The change in 35f4ef2 adds a namespace to the exported target:
https://github.com/AprilRobotics/apriltag/blob/35f4ef2825745a08ec4690ce8c15f8dc903c2868/CMakeLists.txt#L54-L62
which breaks projects that import that target via `target_link_libraries(${PROJECT_NAME} apriltag)`.

To include that namespace, this has to be changed to `target_link_libraries(${PROJECT_NAME} apriltag::apriltag)`.

AndreV84, 2021-05-04:
@christianrauch how to install apriltags so that ROS2 wrapper finds it?
```
--- stderr: apriltag_ros                         
CMake Error at CMakeLists.txt:24 (add_library):
  Target "AprilTagNode" links to target "apriltag::apriltag" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

---
Failed   <<< apriltag_ros [1.67s, exited with code 1]

wxmerkt, 2021-05-04:
I believe target `apriltag::apriltag` does currently not exist. Can you try linking to `apriltag`? We may need to create an ALIAS.

christianrauch, 2021-05-04:
@AndreV84 What did you try so far?

Ideally, `rosdep` resolves all the binary dependencies: `rosdep install --from-paths src -i`.

But I think since the namespace was added, there was no new binary release. So you have to clone https://github.com/AprilRobotics/apriltag manually into your workspace and build everything from source.

AndreV84, 2021-05-04:
@christianrauch Thank you for your response!
So far I tried:
1. cloning both https://github.com/AprilRobotics/apriltag https://github.com/AprilRobotics/apriltag-ros to catkin_ws/src
```
:~/catkin_ws_dashing$ cd apriltag/
@linux:~/catkin_ws_dashing/apriltag$ mkdir build
@linux:~/catkin_ws_dashing/apriltag$ cd build 
@linux:~/catkin_ws_dashing/apriltag/build$ cmake ..
@linux:~/catkin_ws_dashing/apriltag/build$ make -j8
@linux:~/catkin_ws_dashing/apriltag/build$ sudo make install
...
-- Installing: /usr/local/include/apriltag/tagStandard41h12.h
-- Installing: /usr/local/include/apriltag/tagStandard52h13.h
-- Installing: /usr/local/lib/libapriltag.so.3.1.0
-- Up-to-date: /usr/local/lib/libapriltag.so.3
-- Up-to-date: /usr/local/lib/libapriltag.so
```
then I would try to build the dashing workspace that has apriltag_msgs apriltag_ros cloned from https://github.com/christianrauch/apriltag_ros

colcon build

Starting >>> apriltag
Starting >>> apriltag_msgs
--- stderr: apriltag                                                                                                                
CMake Error: The source directory "/home/agx002/catkin_ws_dashing/src/apriltag" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** [cmake_check_build_system] Error 1
---
Failed   <<< apriltag [0.47s, exited with code 2]
Aborted  <<< realsense2_camera_msgs [0.71s]                                                                             
Aborted  <<< apriltag_msgs [0.69s]

Summary: 0 packages finished [1.22s]
  1 package failed: apriltag
  1 package had stderr output: apriltag
  3 packages not processed

```
as apriltags_msgs requires the apriltag folder in the src I moved it to there
```
but now apriltag_ros won't find
```
Starting >>> apriltag_ros
--- stderr: apriltag_ros                          
CMake Error at CMakeLists.txt:24 (add_library):
  Target "AprilTagNode" links to target "apriltag::apriltag" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

---
Failed   <<< apriltag_ros [1.59s, exited with code 1]

Summary: 5 packages finished [12.9s]
  1 package failed: apriltag_ros
  1 package had stderr output: apriltag_ros
```

AndreV84, 2021-05-04:
another error that comes out if to delete all other packages from src but apriltag_ros/ apriltag 
```
- stderr: apriltag_ros                             
In file included from /home/agx002/catkin_ws_dashing/src/apriltag_ros/src/AprilTagNode.cpp:1:0:
/home/agx002/catkin_ws_dashing/src/apriltag_ros/include/AprilTagNode.hpp:11:10: fatal error: image_transport/camera_subscriber.hpp: No such file or directory
 #include <image_transport/camera_subscriber.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
```
locate camera_subscriber
/opt/ros/dashing/include/image_transport/camera_subscriber.h
/opt/ros/melodic/include/image_transport/camera_subscriber.h
```
there seem no hpp file

christianrauch, 2021-05-04:
It looks like you mixed up your workspace with different ROS distributions (`melodic` is ROS1, `dashing` is ROS2).

There is an independent library called `apriltag` at https://github.com/AprilRobotics/apriltag, which works without ROS. The ROS1 node is at https://github.com/AprilRobotics/apriltag_ros, and the ROS2 node is at https://github.com/christianrauch/apriltag_ros. Both ROS nodes are called `apriltag_ros` as they live in different groups and are supposed to be used in separate workspaces, e.g. one for ROS1 and one for ROS2. You should not mix both nodes in the same workspace.

If you want to build the ROS2 workspace, you need the library from https://github.com/AprilRobotics/apriltag, the message definitions from https://github.com/christianrauch/apriltag_msgs and the ROS2 node from https://github.com/christianrauch/apriltag_ros.

AndreV84, 2021-05-05:
@christianrauch 
Thank you for following up!
Exactly! 
ROS1 works fine with default apriltag_ros from https://github.com/AprilRobotics/apriltag_ros also with https://github.com/AprilRobotics/apriltag
But ROS2 Dashing seems have some issues.
Look:
 ```
linux:~$ mkdir ws_dashing
@linux:~$ cd ws_dashing
@linux:~/ws_dashing$ mkdir src
@linux:~/ws_dashing$ cd src
@linux:~/ws_dashing/src$ 
@linux:~/ws_dashing/src$ git clone https://github.com/christianrauch/apriltag_ros
@linux:~/ws_dashing/src$ git clone https://github.com/christianrauch/apriltag_msgs

Unpacking objects: 100% (35/35), done.
@linux:~/ws_dashing/src$ cd ..
@linux:~/ws_dashing$ git clone https://github.com/AprilRobotics/apriltag
Cloning into 'apriltag'...
@linux:~/ws_dashing$ cd apriltag/
@linux:~/ws_dashing/apriltag$ mkdir build
@linux:~/ws_dashing/apriltag$ cd build
@linux:~/ws_dashing/apriltag/build$ cmake ..
-- The C compiler identification is GNU 7.5.0-- Generating done
-- Build files have been written to: /home/void/ws_dashing/apriltag/build
@linux:~/ws_dashing/apriltag/build$ sudo make install
```
at this point it should work as the mixing of ROS yet did not occur, right? but the output is as follows:
```

~/ws_dashing$ colcon build
Starting >>> apriltag
Starting >>> apriltag_msgs
Finished <<< apriltag [14.5s]                                                                       
Finished <<< apriltag_msgs [18.0s]                         
Starting >>> apriltag_ros
--- stderr: apriltag_ros                              
In file included from /home/ws_dashing/src/apriltag_ros/src/AprilTagNode.cpp:1:0:
/home/ws_dashing/src/apriltag_ros/include/AprilTagNode.hpp:11:10: fatal error: image_transport/camera_subscriber.hpp: No such file or directory
 #include <image_transport/camera_subscriber.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/AprilTagNode.dir/src/AprilTagNode.cpp.o] Error 1
make[1]: *** [CMakeFiles/AprilTagNode.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< apriltag_ros [4.89s, exited with code 2]
                                 
Summary: 2 packages finished [23.3s]
  1 package failed: apriltag_ros
  1 package had stderr output: apriltag_ros

```
I tried to look for that specific file but only found 

 ```
/opt/ros/dashing/include/image_transport/camera_subscriber.h
```

AndreV84, 2021-05-05:
@christianrauch 
resolved with
```
sudo bash -c 'echo "#include \"image_transport.h\"" > /opt/ros/dashing/include/image_transport/image_transport.hpp'
sudo bash -c 'echo "#include \"camera_subscriber.h\"" > /opt/ros/dashing/include/image_transport/camera_subscriber.hpp'
```
suggested patch from https://github.com/NVIDIA-AI-IOT/ros2-nvapriltags/issues/7#issuecomment-820750262

christianrauch, 2021-05-05:
@AndreV84 I would recommend that you clone and build all packages as part of your ROS2 workspace via colcon. You manually installed the `apriltag` via `make install`, which works because it's a generic CMake package. But if you build and install it this way system-wide, it does not have to be part of your workspace.

The issue with the headers that you are facing is resulting from an API change in between ROS2 distributions (`dashing` and `foxy`) See the change at https://github.com/christianrauch/apriltag_ros/commit/d8d533406995c46d2de30de6401d61c052c6e3ca. On `master` branch is usually compatible with the latest ROS2 release (`foxy` at the moment). If you have to use an older release (e.g. `dashing`), you have to use a different branch, e.g. https://github.com/christianrauch/apriltag_ros/tree/dashing.

AndreV84, 2021-05-05:
Thank you very much!

commit 35f4ef28 (2020-07-10): Fix transitive dependencies. Fixes #78