AprilRobotics/apriltag #79: Dear ApriltagRobotics team

vigisa93, 2020-03-25:


vigisa93, 2020-03-25:
Dear Apriltag Robotics team
I am a student and I am currently having a bot (build with Raspberry Pie) which has many functionalities. One among them being decoding different symbols (Bar codes, QRCodes etc), Due to the recent advent of April tags, I am fancied to use the same in my robot. I have used OpenCV and zbar to detect and decode QR tags. Can you suggest me a way to decode the April tags as well and how to use your library in the same ?

xiongdahua555, 2020-05-23:
compile install and using

mkrogius, 2020-07-10:
Please follow the instructions on this github for installing and using the library. If you have further, more specific questions, feel free to post a follow up issue.

closed 2020-07-10 by mkrogius