AprilRobotics/apriltag #80: no pose estimates

sidpatki, 2020-03-25:
Hi
I am trying to extract pose information from the detected tags according to the instructions provided in the README file. ```estimate_tag_pose``` returns '0' err but the translation vector in returned pose estimate has all zeros in it. Following is my code snippet and output:

```
 // Draw detection outlines and estimate tag pose
  for(int i = 0; i < zarray_size(detections); i++){
    apriltag_detection_t *det;
    zarray_get(detections, i, &det);

    // Drawing the detection
    line(_detection_image, cv::Point(det->p[0][0], det->p[0][1]),
              cv::Point(det->p[1][0], det->p[1][1]),
              cv::Scalar(0, 0xff, 0), 2);
    line(_detection_image, cv::Point(det->p[0][0], det->p[0][1]),
              cv::Point(det->p[3][0], det->p[3][1]),
              cv::Scalar(0, 0, 0xff), 2);
    line(_detection_image, cv::Point(det->p[1][0], det->p[1][1]),
              cv::Point(det->p[2][0], det->p[2][1]),
              cv::Scalar(0xff, 0, 0), 2);
    line(_detection_image, cv::Point(det->p[2][0], det->p[2][1]),
              cv::Point(det->p[3][0], det->p[3][1]),
              cv::Scalar(0xff, 0, 0), 2);

    stringstream ss;
    ss << det->id;
    cv::String text = ss.str();
    int fontface = cv::FONT_HERSHEY_SCRIPT_SIMPLEX;
    double fontscale = 1.0;
    int baseline;
    cv::Size textsize = cv::getTextSize(text, fontface, fontscale, 2,
                                    &baseline);
    cv::putText(_detection_image, text, cv::Point(det->c[0]-textsize.width/2,
                 det->c[1]+textsize.height/2),
                 fontface, fontscale, cv::Scalar(0xff, 0x99, 0), 2);

  // SINCE I CAN SEE THE DETECTION WHEN I VISUALIZE THE IMAGE, I AM SURE THAT AT THIS POINT THE det CONTAINS VALID DATA.

    // Extraction tag pose
    apriltag_detection_info_t info;
    info.tagsize = size(); // meters
    info.fx = _camera_info.fx(); // pixels
    info.fy = _camera_info.fy(); // pixels
    info.cx = _camera_info.cx(); // pixels
    info.cy = _camera_info.cy(); // pixels
    info.det = det;

   // I HAVE CHECKED THAT info CONTAINS fx, fy,cx,cy, and tagsize values

    // call estimate_tag_pose.
    apriltag_pose_t pose;
    double err = estimate_tag_pose(&info, &pose);
    // Do something with pose.

    cout << "\nerror: " << err << endl;
    cout << "pose.t->nrows: " << pose.t->nrows << "pose.t->ncols: " << pose.t->ncols << endl;
    cout << "pose.R->nrows: " << pose.R->nrows << "pose.R->ncols: " << pose.R->ncols << endl;
    cout << "x: " << pose.t->data[0] << " y: " << pose.t->data[1] << " z: " << pose.t->data[2] << endl;

  }
```

Here is my output :

1 tags detected

error: 0
pose.t->nrows: 3pose.t->ncols: 1
pose.R->nrows: 3pose.R->ncols: 3
x: 0 y: 0 z: 0

It would be great if you could shed light on whats wrong.

xiongdahua555, 2020-05-23:
you should do detect quads and then compute pose。

mkrogius, 2020-07-10:
Can you upload the image you are having this issue with? I don't see any obvious issues with your code.

christian-rauch, 2022-06-28:
Closing this since there is no data to reproduce.

closed 2022-06-28 by christian-rauch