AprilRobotics/apriltag #81: tuning apriltag_quad_thresh_params

maorp, 2020-03-31:
Hey, im struglling with detection of frames that look like should've been easy for detection.
im thinking about playing with apriltag_quad_thresh_params but cant find any example for that.
i can upload the debug images if anyone wants a look.
currently using:
	apriltag_detector_t *td = apriltag_detector_create();
	apriltag_detector_add_family(td, tf);
	td->quad_decimate = 1.f;
	td->quad_sigma = 0.5f;
	td->nthreads = 1;
	td->refine_edges = 0;

[image]

xiongdahua555, 2020-05-23:
I think the problem is：td->quad_sigma =0.5，and your image is blurry。

mkrogius, 2020-07-10:
The problem here is that the tags have been cut out/printed incorrectly so that in the image you have uploaded, the border is ~1pixel wide, and it is blurred with the neighboring pixels such that it does not show up as white.. The outer white border should be cut out with the same width as the black border.

closed 2020-07-10 by mkrogius