AprilRobotics/apriltag #84: AprilTag 3 for Stereo Calibration

ConnorM17, 2020-05-29:
Has there been any work done on implementing stereo calibration between two cameras for extracting intrinsic/extrinsic parameters using AprilTag 3? I was interested in using a board of AprilTags to perform my camera calibration, similar to chessboard or ChArUco calibration. I haven't been able to find any recent work on the AprilTag CameraSuite.

mkrogius, 2020-06-16:
We don't have anything built in, but maybe it would be possible to take the matched features (corner points) from the AprilTag detector and then perform the calibration optimization using that data? The opencv method calibrateCamera looks promising for this at first glance although I have never used it.

yusufguleray, 2021-08-09:
Actually, you can use AprilTags as the markers on the chessboard for the Charuco. Tag families 16h5, 25h9, 36h10 and 36h11 can be used for the markers inside of the Charuco. For more information [Charuco](https://docs.opencv.org/4.5.2/d0/d3c/classcv_1_1aruco_1_1CharucoBoard.html#details) and [OpenCV tag dictionaries](https://docs.opencv.org/4.5.2/d9/d6a/group__aruco.html#gac84398a9ed9dd01306592dd616c2c975).

closed 2024-02-05 by christian-rauch