AprilRobotics/apriltag #85: Estimate_pose stops for a couple of minutes and than returns

vladLozkin, 2020-05-31:
Hello There, 
First of all, thank you for a great job!

I'm having some difficulties with 
```
estimate_tag_pose()
```
simpelfied version of my  code:
```
    apriltag_pose_t pose;
    apriltag_detection_info_t d_info;
    d_info.tagsize = tagsize;
    d_info.fx = focal;
    d_info.fy = focal;
    d_info.cx = cx;
    d_info.cy = cy;
    
    while (// cv::cap is not empty)
    {
         apriltag_detection_t *det;
         zarray_get(detections, i, &det);
         d_info.det = det;
         double err = estimate_tag_pose(&d_info, &pose);
    }
    
```
the estimation works great when I get back the results.
At a certain point, estimate_tag_pose() does not return and when I debugged the source code I see that
it stops inside: 
```
orthogonal_iteration() 
```
the specific line where there is get stuck is:
```
matd_svd_t M3_svd = matd_svd(M3);
```

I'm using an 800 mm april tag
36h11
id 42

this is the frame it stops on 
[image]

Any idea will be grate

alddiaz, 2020-05-31:
Your implementation looks pretty OK.
Could you please share a full resolution PNG image to double check?

vladLozkin, 2020-05-31:
I've added the specific frame and the full video to this drive folder
https://drive.google.com/drive/folders/1eCblBHvMo3T61I-ZqAoT1h0raGa6uI1P?usp=sharing

notice that it happens always on the same frame.
after about 3,4 minutes, estimate_tag_pose() returns and then after a couple of frames it happens again, and so on...

alddiaz, 2020-05-31:
> 800 mm april tag 36h11 id 42 this is the frame it stops on [image: tag

Hej,

I've just tested on the conflict image and got a positive result. I am
using my own MATLAB wrapper implementation that you can download at:
https://github.com/alddiaz/MATLAB_AprilTag3

So, to me it seems that you code may be a little fishy. Few things you can
check are clearing up the memory of the pointer variables you allocate
(otherwise you can cause a memory leakage).

BTW, nice drone flying skills ;)

Here you are the results:
>> Tag=apriltags(Ig,.8,[2701 0 size(I,2)/2;0 2701 size(I,1);0 0 1])
Tag =
  struct with fields:
         id: 42
    hamming: 0
     margin: 169.5241
          H: [3×3 double]
     center: [2×1 double]
    corners: [2×4 double]
          p: [3×1 double]
          R: [3×3 double]
>> Tag.p
ans =
   -4.2451
  -18.9710
   79.1889
>> Tag.R
ans =
    0.9528   -0.2805   -0.1160
    0.2054    0.8771   -0.4341
    0.2235    0.3898    0.8934

*Aldo Díaz*, PhD
School of Mechanical Engineering
University of Campinas, Brazil

vladLozkin, 2020-06-02:
Hi 
Thanks for the check alddiaz!

I'll try looking at my code once again.
I believe it's a memory leakage issue because it stops always on the same frame, no matter what the state of the machine and additionally I already checked that options with different tools.
alddiaz if you want I can teach you how to fly it :)
maybe you know somebody with c++ implementation that can check it also?

**Can anybody with c++ implementation check the video in this folder?
https://drive.google.com/drive/folders/1eCblBHvMo3T61I-ZqAoT1h0raGa6uI1P?usp=sharing**

Thnx in advance

alddiaz, 2020-06-02:
Hej again,

Despite my wrapper is for MATLAB, it is actually C++ code I wrote. So, take
a look at my code to see how I did to clear up pointers and don't forge to
report back when you solve the problem.
Lessons learned are important part of the process ;)
https://github.com/alddiaz/MATLAB_AprilTag3/blob/master/Machine%20Vision%20Toolbox%20for%20MATLAB/vision/mex/apriltags.c

P.D.: Looking forward for those free drone flight/crash lessons.

Regards,

*Aldo Díaz*, PhD
School of Mechanical Engineering
University of Campinas, Brazil

vladLozkin, 2020-06-02:
I'm embarrassed to write this comment :'(
Eventually, the problem was that my tag size was in mm and not in meters.
800 meters is a lot more than 0.8 meters....

You do a great job with this repo,
Thank you for the time, the patience, tries and the explanations!

alddiaz, 2020-06-02:
Your`re welcome Vlad.
I am glad to know you manage to solve the problem, so this is good news :)

Regards,

*Aldo Díaz*, PhD
School of Mechanical Engineering
University of Campinas, Brazil

closed 2020-06-04 by vladLozkin