AprilRobotics/apriltag #87: release to noetic and foxy

christian-rauch, 2020-06-12:
@wxmerkt Could we release the library to `noetic` (ROS1) and `foxy` (ROS2)?

Since `noetic` will be the last ROS1 version ever and `foxy` is supported for 3 years, I am not going to ask for further releases for a while after this :-)

wxmerkt, 2020-06-12:
Absolutely, happy to make the release.

@mkrogius, do we want to tag a new release since there have been 29 commits to master since the last version tag 3.1.1? This also requires an update of the version number in [package.xml](https://github.com/AprilRobotics/apriltag/blob/cad1009f9fbf457e2a5393c0fb6c8a05042a0611/package.xml#L5).

mkrogius, 2020-06-12:
Done, bumped to 3.1.2

wxmerkt, 2020-06-13:
Thank you for the swift action! I noticed the changes are more than bugfixes as also a Python3 wrapper was added. 

I've made the releases and will monitor what comes back from the buildfarm. I feel we may have to do patches or fixes for GNUInstallDirs or due to the addition of the Python wrapper.

Here are the rosdistro PRs:
- Noetic: https://github.com/ros/rosdistro/pull/25455
- Foxy: https://github.com/ros/rosdistro/pull/25456

Please note these releases of 3.1.2 do not include the Python wrapper.

https://github.com/AprilRobotics/apriltag/pull/88 fixes the dependency on NumPy which will then in turn build the Python wrappers in the next release.

wxmerkt, 2020-06-16:
Builds on both Noetic and Foxy are now passing - we can close this issue.

christian-rauch, 2020-06-16:
Thanks @wxmerkt !

closed 2020-06-16 by christian-rauch