AprilRobotics/apriltag #90: Obtaining euler angles from apriltag_pose_t?

FROGbots-4634, 2020-06-21:
Once I have obtained an `apriltag_pose_t`, how can I get the euler angles from the rotation matrix provided at `pose.R`?

mkrogius, 2020-07-10:
You can use the method common/doubles_mat_to_quat to get a quaternion from the matrix. Then you can use common/doubles_quat_to_rpy to convert that quaternion to RPY.

closed 2020-07-10 by mkrogius

vladLozkin, 2020-07-19:
Hi, mkrogius,
Im looking for obtaining euler angles from post also but I cannot find the functions that you mentioned in the code...

can you point to a more specific location / show how to call them from cpp?

vladLozkin, 2020-07-19:
> You can use the method common/doubles_mat_to_quat to get a quaternion from the matrix. Then you can use common/doubles_quat_to_rpy to convert that quaternion to RPY.

Hi, mkrogius,
Im looking for obtaining euler angles from post also but I cannot find the functions that you mentioned in the code...

can you point to a more specific location / show how to call them from cpp?

mkrogius, 2020-07-20:
The methods are declared in this header file: https://github.com/AprilRobotics/apriltag/blob/master/common/doubles.h

which templates the code from this header file: https://github.com/AprilRobotics/apriltag/blob/master/common/doubles_floats_impl.h#L456

chobitsfan, 2022-09-28:
doubles_mat_to_quat assumes a 4x4 matrix, but pose.R is a 3x3 matrix. You need to re-arrange pose.R before give it to doubles_mat_to_quat