AprilRobotics/apriltag #91: Estimating pose of Apriltags via Spherical/360 camera

swapnilsayansaha, 2020-06-25:
I was wondering if it's possible to detect Apriltags using stitched images from spherical/360 cameras, as the camera intrinsic parameters are of different nature. Example image from 360 camera:

[image]

mkrogius, 2020-07-10:
It might be possible, especially when the tag is further away from the camera so that it is less distorted. One detector parameter you can try changing is qtp.max_line_fit_mse which should basically help detecting lines that are not very straight. If you do test out the detection performance, please post the results here!

swapnilsayansaha, 2020-07-10:
Sure I will let you know. I am using DuckieTag's bindings.

kin0201, 2021-10-17:
Hi @swapnilsayansaha Any luck on estimating the pose from a 360 image? I come across difficulties when determining what intrinsic camera parameters to use as well. Would appreciate it if you can share any experience on that.