AprilRobotics/apriltag #93: opencv2 missing

opticalCoherence, 2020-06-29:
Hello,

I am trying to get the example applications running and am hitting a snag. I installed OpenCV, and all the other packages, but when I do: make opencv_demo

I get the following error: 

make opencv_demo
   opencv_demo.o
opencv_demo.cc:30:10: fatal error: opencv2/opencv.hpp: No such file or directory
   30 | #include "opencv2/opencv.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:28: opencv_demo.o] Error 1

This makes me think that I am missing that particular .hpp file?

Sorry if this doesn't make much sense, I am pretty new to both Linux and C. Thanks!

mkrogius, 2020-08-14:
It sounds like its having trouble finding your opencv installation. Can you try building with cmake for a better error message? In your root directory type:
    cmake .
    sudo make install

brmarkus, 2020-08-14:
I'm using AprilTagv3 in an OpenVINO environment, i.e. it has OpenCV installed plus CMAKE related variables for resolving the dependencies (like libraries, library-search-path, header-include-search-path).
How have you "installed OpenCV, and all the other packages"?

closed 2022-03-10 by wxmerkt

Lucca-cherries, 2024-01-11:
> It sounds like its having trouble finding your opencv installation. Can you try building with cmake for a better error message? In your root directory type: cmake . sudo make install

I am having the same problem and this two lines of commands work! Thanks @mkrogius.