AprilRobotics/apriltag #96: Speedup detection specifying tag size boundaries

svetliy95, 2020-07-08:
In my application, the size of tags in an image do not vary so much. Is there a way to specify boundaries for the tag size in pixels, so it can speed up the detection? I know that ArUco has such feature.
Image decimation is not an option for me, because I need the pose estimation to be exact as possible.

mkrogius, 2020-07-14:
The equivalent parameter would be the quad_decimate parameter. Set this higher in order to speed up detection at the cost of missing smaller tags if you set it above 2. This will not affect the accuracy of the tag corners since it decimates the image for only the detection part of the algorithm.

closed 2020-07-14 by mkrogius

SLAMgreen, 2021-09-13:
> The equivalent parameter would be the quad_decimate parameter. Set this higher in order to speed up detection at the cost of missing smaller tags if you set it above 2. This will not affect the accuracy of the tag corners since it decimates the image for only the detection part of the algorithm.

Do you test about the detecting result when set the param above 2?

mkrogius, 2021-09-15:
Yes, see the paper: https://april.eecs.umich.edu/papers/details.php?name=krogius2019iros

It will give greater speedups, but at the cost of detection distance. The alternative option would be to downsample the image yourself before passing it into apriltag and leave quad_decimate at 2. This could be even slightly faster, at the cost of slightly less precision in the corner locations.