AprilRobotics/apriltag #97: CMake 3.10 or higher is required.  You are running version 3.5.1

Rayman, 2020-07-15:
I noticed that the CMake version that is required has been upped in this commit: 35f4ef2825745a08ec4690ce8c15f8dc903c286835f4ef2825745a08ec4690ce8c15f8dc903c2868

Kinetic doesn't have this version yet. Are you sure you want to drop kinetic?
```
_______________________________________________________________________________
Errors << apriltag:cmake /project/logs/apriltag/build.cmake.000.log
CMake Error at /project/src/apriltag/CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.10 or higher is required.  You are running version 3.5.1
cd /project/build/apriltag; catkin build --get-env apriltag | catkin env -si  /usr/bin/cmake /project/src/apriltag --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/project/devel; cd -
...............................................................................
Failed << apriltag:cmake                                          [ Exited with code 1 ]
```

marcelino-pensa, 2020-07-16:
Same happened here... seems like you can fix it by compiling the tag `3.1.2` (`git checkout 3.1.2`) if you don't need CI integration in ROS `noetic` or `foxy`.
You'd have to update CMake otherwise

mkrogius, 2020-07-16:
I'll put it back to the previous version, didn't realize it caused problems with ROS

closed 2020-07-16 by mkrogius

marcelino-pensa, 2020-07-16:
Thanks, @mkrogius! This is not really a ROS problem, but rather an Ubuntu problem... Ubuntu 16.04 natively installs `3.5.1`, and can install `3.10` through PPA. If you really have to reason to update to `3.10`, then keeping it at `3.5.1` will maintain wider backward compatibility! Thanks a lot!

Rayman, 2020-07-17:
Thanks everyone!

commit e24308a2 (2020-07-16): Revert Cmake version increase. Fixes #97