AprilRobotics/apriltag #99: Update release repository for ROS 2 rolling release support

nuclearsandwich, 2020-07-23:
In order to be automatically released from Foxy to Rolling, and in order automatically release from Rolling to future distributions, the rolling release automation needs to be able to push to release repositories.
For this reason, [REP 2002](https://www.ros.org/reps/rep-2002.html#managing-build-farm-writable-release-repositories) specified that release repositories for the Rolling distribution would be created in the [ros2-gbp](https://github.com/ros2-gbp) organization if they were not part of that organization already.

When the Rolling distribution was created, a new release repository for apriltag, was created at https://github.com/ros2-gbp/apriltag-release

My recommendation would be for the apriltag team to switch to using the ros2-gbp release repository for all ROS 2 releases of apriltag.
With your confirmation, I can add maintainers to the repository in ros2-gbp with full administrative access. Then we would only need to update the release repository entry in other distros where apriltag is released.

If you wish, you may keep using the existing release repository for ROS 1 releases or you could use the ros2-gbp repository for both ROS 1 and ROS 2 releases and archive the previous release repository as-is, keeping it intact for any sources still using it for earlier releases.

wxmerkt, 2022-03-10:
Thank you for creating this - Galactic and Rolling are now released from the new ROS2 release repository, with Foxy still being on the original release repository. I don't think that should cause any issues, so if it's okay with you, can we continue to use the existing release repository for Foxy?

nuclearsandwich, 2022-05-12:
> I don't think that should cause any issues, so if it's okay with you, can we continue to use the existing release repository for Foxy?

With apologies for the tardy answer (my non @-mentions take a while to get to) yes, if that's your preference it works for me. Using ros2-gbp for all ROS 2 distros is recommended just to avoid confusion but the only strong recommendation is to use the ros2-gbp repository for all distros which have been touched by the migration tool so that all their releases occur in the same release repository and that list exactly every ROS 2 distro _after_ foxy.

wxmerkt, 2022-05-12:
@nuclearsandwich Perfect, then we are in agreement -- foxy stays on apriltag/apriltag-release and galactic, rolling, humble, and newer uses ros2-gbp/apriltag-release as released :-)

closed 2022-05-12 by wxmerkt