AprilRobotics/apriltag #100: No rule to make target '../libapriltag.a' when making demos

WaterFriend, 2020-08-14:
Hi Everybody:
I am using the Ubuntu18.04 virtual machine (virtual box) as my environment. I have successfully installed OpenCV 3.4.8 and Apriltag library. But when I try to cmake the demo by following the official instructions "_make opencv_demo_" and "_make apriltag_demo_", the following errors show up:

```
$ cd example/
$ make opencv_demo 
make: *** No rule to make target '../libapriltag.a', needed by 'opencv_demo'.  Stop.
$ make apriltag_demo 
make: *** No rule to make target '../libapriltag.a', needed by 'apriltag_demo'.  Stop.
```

I try to ignore the error and directly run the demos, the following errors show up:

```
$ ./apriltag_demo 
apriltag_demo: error while loading shared libraries: libapriltag.so.3: cannot open shared object file: No such file or directory
$ ./opencv_demo
opencv_demo: error while loading shared libraries: libapriltag.so.3: cannot open shared object file: No such file or directory
```

But in fact, I have libapriltag.so.3 installed in /usr/local/lib as Apriltag library is successfully installed. Is anyone has any idea with these issues? Thanks for any advice

jeremi, 2020-09-05:
@WaterFriend I had the same issue, running `sudo ldconfig` resolved it.

Reihold-Chen, 2021-08-02:
I meet the same mistake, did you have solved it ?

Zhang-Qinghui, 2021-11-17:
I meet the same mistake, did you have solved it ?

christian-rauch, 2021-11-17:
@Reihold-Chen @Zhang-Qinghui Did running `sudo ldconfig` work for you?

Are you using `cmake` or `make`? The documentation has instructions for both. You can also try to run CMake with Ninja instead:
```
cmake . -GNinja
ninja
```

Hola39e, 2021-12-14:
I met the same mistake and got a solution. 
the `../libapriltag.a` file doesnt exsit in the right path when using `cmake` to build apriltag library, the lib was directly added to `/usr/local/lib`.
so change the `Makefile` file in /example,  delete `../libapriltag.a` text and add `pkg-config --libs apriltag` instead, then is could be compiled successfully.

```
apriltag_demo: apriltag_demo.o  
      @echo "   [$@]"
      @$(CC) -o $@ $^ $(LDFLAGS) `pkg-config --libs apriltag`
 
opencv_demo: opencv_demo.o
     @echo "   [$@]"
     @$(CXX) -o $@ $^ $(LDFLAGS) `pkg-config --libs opencv` `pkg-config --libs apriltag`
```

christian-rauch, 2021-12-14:
> I met the same mistake and got a solution.

What did you do?

The `cmake` instructions on the main page work for me, I can run `./opencv_demo` afterwards. The `make` file exists in parallel to the `cmake` system. But don't use them in parallel since running `cmake` in the source folder will override the Makefile.

Hola39e, 2021-12-14:
I get what's your meaning. Thanks. What I did was:
```
$cmake .
$sudo make install
$cd ./example
$make all
make: *** No rule to make target '../libapriltag.a', needed by 'apriltag_demo'.  Stop.
```
After I used `cmake` method to install, I didn't find static lib `libapriltag.a` in the source code dir, that's why I cannot run `make all`  instructions in `/example`. 
I just tried to use another way `make` + `sudo make install` to install in another virtual environment, then I found `make all` in `/example` works.
Thanks for helping, I have solved my problem.

christian-rauch, 2021-12-14:
Yes, don't mix the build systems.

The Makefile already builds the example, it prints: `make[1]: Entering directory '/tmp/apriltag/example'`.

If you need the static library, you can either use the Makefile as per the README instructions, or you use cmake with the `BUILD_SHARED_LIBS` set to `OFF` (introduced in https://github.com/AprilRobotics/apriltag/pull/172):
```
cmake . -DBUILD_SHARED_LIBS=OFF
make -j
```

christian-rauch, 2022-01-27:
The documentation has been updated.

closed 2022-01-27 by christian-rauch