AprilRobotics/apriltag #104: Major bug in tag pose estimation

cstathis, 2020-08-20:
`orthogonal_iteration` can sometimes return an R matrix that contains a reflection due to sign ambiguity in SVD. In my experience this led to an invalid orientation being returned by `estimate_tag_pose_orthogonal_iteration` and also causes it to fail to find the second pose (it will print "Error, more than one minimum found.").

This problem can be solved by checking the determinant of the R matrix in `orthogonal_iteration`, and if it is -1 instead of 1, flipping the sign on the 3rd column.

As an aside, the performance of this function could also be greatly improved by adding a convergence criteria rather than always running for a fixed number of iterations.

jerry73204, 2020-10-17:
I can confirm this issue by running `pose_estimation()` on the image in the [link](https://robotics.shanghaitech.edu.cn/node/201). It produces the rotation matrix. You can check the determinant is roughly -0.9999999999999997.

```
[[-5.59800153e-01, -8.28625514e-01, -1.88294478e-03],
 [-8.28624997e-01,  5.59802668e-01, -1.26020952e-03],
 [-2.09831927e-03, -8.54789624e-04,  9.99997433e-01]]
```

Another issue is that is specific input parameters cause runtime segfaut, for example, fx=0.000001 fy=1.0. I observe it does not handle small floating points correctly. (<= 1e-4)

(Just amazed that the big name april lab never leave any comment since this serious bug opens months.)

ShrutheeshIR, 2020-10-23:
Hey @cstathis I'm facing the same issue. Could you share the code snippet to correct for the wrong pose?

mkrogius, 2021-01-07:
I was kind of surprised to see that this was causing problems, so I went back and checked the pose estimation code against my dataset. I found zero examples where the determinant ended up as -1. That said, the theoretical argument that this should be fixed is strong enough that I have gone ahead and made a tentative fix in f409268c918ff0b05aa495628160c133baeedfb7

I suspect this problem may only actually show up in cases where the data is not clean, e.g. the tag does not lie flat or the camera calibration is incorrect.

@cstathis would you mind sending me any data you have of cases where pose estimation fails in this manner?

@jerry73204 What camera parameters did you use in order to reproduce the issue with the image you linked? It looks like the tag does not lie flat in that image, which could potentially cause this sort of issue.

Also, I should note that while AprilTag may be widely used, there is no funding of any sort to support it. There is only me, a single PhD student, who fixes bugs in his free time. I should also not that I would not consider this a major bug, because a) it does not affect the core functionality of tag detection and b) it does not affect a broad set of users (as far as I know).

closed 2021-06-15 by mkrogius