AprilRobotics/apriltag #107: gcc: error: libpython3.7m.a: No such file or directory

pra-dan, 2020-09-22:
I am using `pyenv` to run multiple python instances on my system. I cloned apriltag into the directory `~/github/apriltag`. The local python in this directory is: 
```
$ python -V
Python 3.7.4
```
I have following versions installed:
```
$ pyenv versions
  system
* 3.7.4 (set by /home/pra-dan/github/apriltag/.python-version)
  3.7.8
  3.7.8/envs/hot
  3.7.8/envs/slam
```

 I performed `$ cmake .` and it ran fine. 

Now when I run `sudo make install`, I get the error 
```console
gcc: error: libpython3.7m.a: No such file or directory
make[2]: *** [CMakeFiles/apriltag_python.dir/build.make:62: apriltag.cpython-37m-x86_64-linux-gnu.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/apriltag_python.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

The entire logs are:
```console
$ sudo make install
[sudo] password for pra-dan: 
[  2%] Building C object CMakeFiles/apriltag.dir/apriltag.c.o
[  5%] Building C object CMakeFiles/apriltag.dir/apriltag_pose.c.o
[  8%] Building C object CMakeFiles/apriltag.dir/apriltag_quad_thresh.c.o
[ 10%] Building C object CMakeFiles/apriltag.dir/common/g2d.c.o
[ 13%] Building C object CMakeFiles/apriltag.dir/common/getopt.c.o
[ 16%] Building C object CMakeFiles/apriltag.dir/common/homography.c.o
[ 18%] Building C object CMakeFiles/apriltag.dir/common/image_u8.c.o
[ 21%] Building C object CMakeFiles/apriltag.dir/common/image_u8x3.c.o
[ 24%] Building C object CMakeFiles/apriltag.dir/common/image_u8x4.c.o
[ 27%] Building C object CMakeFiles/apriltag.dir/common/matd.c.o
[ 29%] Building C object CMakeFiles/apriltag.dir/common/pam.c.o
[ 32%] Building C object CMakeFiles/apriltag.dir/common/pjpeg-idct.c.o
[ 35%] Building C object CMakeFiles/apriltag.dir/common/pjpeg.c.o
[ 37%] Building C object CMakeFiles/apriltag.dir/common/pnm.c.o
[ 40%] Building C object CMakeFiles/apriltag.dir/common/string_util.c.o
[ 43%] Building C object CMakeFiles/apriltag.dir/common/svd22.c.o
[ 45%] Building C object CMakeFiles/apriltag.dir/common/time_util.c.o
[ 48%] Building C object CMakeFiles/apriltag.dir/common/unionfind.c.o
[ 51%] Building C object CMakeFiles/apriltag.dir/common/workerpool.c.o
[ 54%] Building C object CMakeFiles/apriltag.dir/common/zarray.c.o
[ 56%] Building C object CMakeFiles/apriltag.dir/common/zhash.c.o
[ 59%] Building C object CMakeFiles/apriltag.dir/common/zmaxheap.c.o
[ 62%] Building C object CMakeFiles/apriltag.dir/tag16h5.c.o
[ 64%] Building C object CMakeFiles/apriltag.dir/tag25h9.c.o
[ 67%] Building C object CMakeFiles/apriltag.dir/tag36h11.c.o
[ 70%] Building C object CMakeFiles/apriltag.dir/tagCircle21h7.c.o
[ 72%] Building C object CMakeFiles/apriltag.dir/tagCircle49h12.c.o
[ 75%] Building C object CMakeFiles/apriltag.dir/tagCustom48h12.c.o
[ 78%] Building C object CMakeFiles/apriltag.dir/tagStandard41h12.c.o
[ 81%] Building C object CMakeFiles/apriltag.dir/tagStandard52h13.c.o
[ 83%] Linking C shared library libapriltag.so
[ 83%] Built target apriltag
[ 86%] Building C object CMakeFiles/apriltag_demo.dir/example/apriltag_demo.c.o
[ 89%] Linking C executable apriltag_demo
[ 89%] Built target apriltag_demo
[ 91%] Generating apriltag_py_type.docstring.h
[ 94%] Generating apriltag_detect.docstring.h
[ 97%] Generating apriltag_pywrap.o
[100%] Generating apriltag.cpython-37m-x86_64-linux-gnu.so
gcc: error: libpython3.7m.a: No such file or directory
make[2]: *** [CMakeFiles/apriltag_python.dir/build.make:62: apriltag.cpython-37m-x86_64-linux-gnu.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/apriltag_python.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

How to proceed ?

iuliuv, 2020-10-02:
try installing the python3-dev package (or the specific version python3.7-dev)?

zzhh00, 2020-12-17:
I have met the same issue on ubuntu16.04.
`gcc: error: libpython3.6m.a: No such file or directory`
`CMakeFiles/apriltag_python.dir/build.make:61: recipe for target 'apriltag.cpython-36m-x86_64-linux-gnu.so' failed`
`make[2]: *** [apriltag.cpython-36m-x86_64-linux-gnu.so] Error 1`
`CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/apriltag_python.dir/all' failed`
`make[1]: *** [CMakeFiles/apriltag_python.dir/all] Error 2`
`Makefile:127: recipe for target 'all' failed`
`make: *** [all] Error 2`
How to deal with it?

pra-dan, 2020-12-17:
I had to resort to https://www.orangenarwhals.com/2018/04/getting-started-with-apriltags-on-ubuntu-16-04/
It worked

zzhh00, 2020-12-17:
I think it is not the same as Apriltag, and the compiling of Aprilrag_ros will fail.

mkrogius, 2021-05-07:
The simple answer here may be to not build the python library. It can be disabled in the build with the variable: https://github.com/AprilRobotics/apriltag/blob/master/CMakeLists.txt#L103

KAIKAIZHANG, 2021-12-06:
> 

i have the same question, have you solved this problem?

KAIKAIZHANG, 2021-12-06:
> I have met the same issue on ubuntu16.04. `gcc: error: libpython3.6m.a: No such file or directory` `CMakeFiles/apriltag_python.dir/build.make:61: recipe for target [...]

i have the same question, have you solved this problem?

christian-rauch, 2021-12-06:
The static library `libpython3.7m.a` should be located at `/usr/lib/x86_64-linux-gnu/libpython3.7m.a` and is provided by package 	`libpython3.7-dev`. This will apply similarly to the 3.6 and 3.8 versions.

closed 2022-01-27 by christian-rauch

krrish94, 2022-08-20:
Adding a `PYTHON_LIBRARY_DIR` variable worked for me. Something like

```
catkin build -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.9 -DPYTHON_LIBRARY_DIR=/usr/lib/x86_64-linux-gnu
```