AprilRobotics/apriltag #108: Significantly slower than the default python 3 module (0.0.16), on ARM

iuliuv, 2020-10-02:
Hi, great library, first of all!!

so far we used the version installed by default by pip in python3 on Jetson Nano (ARM Cortex A57):
https://pypi.org/project/apriltag/
It takes about 30ms to process a frame.

We uninstalled it and compiled/installed the version on your repository, using the provided commands. The result is significantly slower (about 500ms per frame). 
Are we missing something? are we doing something wrong? Any clues? 

both version use about 100% of the CPU (one core)

we would like to use your version for the custom tags.

thanks,
Iuliu.

mkrogius, 2020-10-02:
That is odd, on x86, using c, I know that this library is approximately 2x the speed of the version you are using. So then the slowdown could be caused by the two different python wrappers or perhaps by some piece of the new code that is slow on the cortex A57. The easiest thing to do is probably to rule out the change of architecture. Can you perform the same tests that you have done, but do them on an x86 system this time?

iuliuv, 2020-10-02:
Hi we ran a simple test on x86, same conditions, calling the two libraries from Python3.
the pip default library runs in aprox 2ms per frame, the apriltag from this repository needs 8ms for the same frames.
we are initializing the detector with: apriltag.apriltag("tag36h11") (we get similar results with the Custom tags that we need)

Any ideas?

the code is very simple:
`import apriltag
import cv2
import time

vc = cv2.VideoCapture(0)

#april = apriltag.Detector()          #when pip library is installed
april = apriltag.apriltag("tag36h11") #when AprilRobotics/apriltag installed

while True:
    succes, frame = vc.read()
    frame = cv2.cvtColor(frame, cv2.COLOR_RGB2GRAY)
    t = time.time()
    for i in range(100):
        result = april.detect(frame)
    print(time.time()-t)
    #print(result)`

iuliuv, 2020-10-02:
We also ran the opencv_demo in your repository, on the same x86 machine, everything compiled on a Ubuntu 18 machine with the default libraries and compiler, modified to time just the call to apriltag_detector_detect(...), and it takes 73ms on average, per frame, which I would say is quite a lot. No command line options were passed to opencv_demo

mkrogius, 2020-10-02:
The opencv_demo was creating debug output by default. I fixed this in a push just now.

I tested myself and did find a similar speed difference between the two libraries. It doesn't have anything to do with the python wrapper, but rather with the default settings for each library. When both use the same settings, AprilTag 3 is almost 2x faster.

The main difference was in thresholding, the older library was being much more aggessive with its thresholding. You can try changing the following settings to improve performance for your application. The tradeoff in these parameters is pretty much always between detecting more tags vs performance, so if you really care about speed, try changing these parameters until your tags are just barely detected.

1) Set quad_thresh_params->min_white_black_diff to a higher value. This will speed up processing at the cost of missing out on low-contrast tags.
2) Set quad_decimate to 2. This will improve speed at the cost of missing a very small number of far away tags.
3) Downscale the image before handing it off to AprilTag. This will improve speed at the cost of missing far away tags. The more you downscale, the greater the effect.
4) Set quad_sigma to a higher value. This will blur the image more. This will improve speed on grainier images at the cost of missing out on lower contrast tags.
5) Increase quad_thresh_params->min_cluster_pixels to a higher value. This will improve speed at the cost of missing some far away tags. You should optimize suggestions 2&3 before tuning this value.

For tuning all of this it can be helpful to look at the debug output to see how the thresholding/clustering/quad fitting operations are going. Ideally you want to have the minimum amount of thresholds/clusters/quads while still detecting your tags.

iuliuv, 2020-10-19:
You were right about opencv_demo - much faster without the debug info!

Would be very helpful to have a summary of settings to obtain the 2x speed-up compared to the previous version, but already much better

closed 2020-11-12 by mkrogius

maharshi114, 2021-06-15:
Hey @iuliuv, Can you explain how you get max performance in Nvidia Jetson nano?  I am facing similar issues.

iuliuv, 2021-06-16:
The latest version has the debug info deactivated by default, so it is already much faster... however we could not get it to be faster than the original in the pip repository. Some parameter tweaking is needed, but we were not sure what to change.. sry...