AprilRobotics/apriltag #109: issues when installing on Mac

florakth, 2020-10-08:
Hi, I met issues when installing on my MacOX system.

1. git clone https://github.com/AprilRobotics/apriltag
2. cd apriltag
3. cmake .
4. sudo make install

Errors come at the last step: 
Scanning dependencies of target apriltag
[  2%] Building C object CMakeFiles/apriltag.dir/apriltag.c.o
[  5%] Building C object CMakeFiles/apriltag.dir/apriltag_pose.c.o
[  7%] Building C object CMakeFiles/apriltag.dir/apriltag_quad_thresh.c.o
[ 10%] Building C object CMakeFiles/apriltag.dir/common/g2d.c.o
[ 12%] Building C object CMakeFiles/apriltag.dir/common/getopt.c.o
[ 15%] Building C object CMakeFiles/apriltag.dir/common/homography.c.o
[ 17%] Building C object CMakeFiles/apriltag.dir/common/image_u8.c.o
[ 20%] Building C object CMakeFiles/apriltag.dir/common/image_u8x3.c.o
[ 23%] Building C object CMakeFiles/apriltag.dir/common/image_u8x4.c.o
[ 25%] Building C object CMakeFiles/apriltag.dir/common/matd.c.o
[ 28%] Building C object CMakeFiles/apriltag.dir/common/pam.c.o
[ 30%] Building C object CMakeFiles/apriltag.dir/common/pjpeg-idct.c.o
[ 33%] Building C object CMakeFiles/apriltag.dir/common/pjpeg.c.o
[ 35%] Building C object CMakeFiles/apriltag.dir/common/pnm.c.o
[ 38%] Building C object CMakeFiles/apriltag.dir/common/string_util.c.o
[ 41%] Building C object CMakeFiles/apriltag.dir/common/svd22.c.o
[ 43%] Building C object CMakeFiles/apriltag.dir/common/time_util.c.o
[ 46%] Building C object CMakeFiles/apriltag.dir/common/unionfind.c.o
[ 48%] Building C object CMakeFiles/apriltag.dir/common/workerpool.c.o
[ 51%] Building C object CMakeFiles/apriltag.dir/common/zarray.c.o
[ 53%] Building C object CMakeFiles/apriltag.dir/common/zhash.c.o
[ 56%] Building C object CMakeFiles/apriltag.dir/common/zmaxheap.c.o
[ 58%] Building C object CMakeFiles/apriltag.dir/tag16h5.c.o
[ 61%] Building C object CMakeFiles/apriltag.dir/tag25h9.c.o
[ 64%] Building C object CMakeFiles/apriltag.dir/tag36h11.c.o
[ 66%] Building C object CMakeFiles/apriltag.dir/tagCircle21h7.c.o
[ 69%] Building C object CMakeFiles/apriltag.dir/tagCircle49h12.c.o
[ 71%] Building C object CMakeFiles/apriltag.dir/tagCustom48h12.c.o
[ 74%] Building C object CMakeFiles/apriltag.dir/tagStandard41h12.c.o
[ 76%] Building C object CMakeFiles/apriltag.dir/tagStandard52h13.c.o
[ 79%] Linking C shared library libapriltag.dylib
[ 79%] Built target apriltag
Scanning dependencies of target apriltag_python
[ 82%] Generating apriltag_py_type.docstring.h
[ 84%] Generating apriltag_detect.docstring.h
[ 87%] Generating apriltag_pywrap.o
[ 89%] Generating apriltag.cpython-37m-darwin.so
clang: error: no such file or directory: '/Users.../apriltag/libapriltag.so'
make[2]: *** [apriltag.cpython-37m-darwin.so] Error 1
make[1]: *** [CMakeFiles/apriltag_python.dir/all] Error 2
make: *** [all] Error 2

Could anyone please give some suggestions on this? Thanks a lot!

mkrogius, 2020-10-08:
The problem seems to be just with the python wrapper. If you don't need to use the python wrapper you should be able to build successfully by commenting out lines 64-95 of CMakeLists.txt and then repeating all the build steps starting with cmake ...

florakth, 2020-10-08:
Hi,

Thank you so much for the reply, I do need to use this library in a python file, so should I remove them?

I tried many ways, but  none of them works. Is there any other requirement for this library?

like visual studio, or are there some different requirements for Mac system?

Thank you again.

Best,

Flora

mkrogius, 2020-10-08:
I just pushed a change to the repo that should fix this issue. Can you re-download and try again?

florakth, 2020-10-08:
Thank you so much ! I redownloaded it and ran again, it seems the same error:
[ 79%] Linking C shared library libapriltag.dylib
[ 79%] Built target apriltag
Scanning dependencies of target apriltag_python
[ 82%] Generating apriltag_py_type.docstring.h
[ 84%] Generating apriltag_detect.docstring.h
[ 87%] Generating apriltag_pywrap.o
[ 89%] Generating apriltag.cpython-37m-darwin.so
ld: unknown option: -rpath=lib
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [apriltag.cpython-37m-darwin.so] Error 1
make[1]: *** [CMakeFiles/apriltag_python.dir/all] Error 2
make: *** [all] Error 2

mkrogius, 2020-10-08:
It looks like its an error from slightly further along in the process, we are making progress here. I made a change that should fix it, try re-downloading again. I think you may be the first person to try compiling on Mac OS ever, since these issues are all Mac specific.

florakth, 2020-10-08:
It is very kind of you! I really appreciate your work!

Now it shows be installed successfully, then when I run the python code, an error comes again:

Traceback (most recent call last):
  File "detect_marker.py", line 3, in <module>
    from apriltag import apriltag
ImportError: cannot import name 'apriltag' from 'apriltag' (/Applications/anaconda3/lib/python3.7/site-packages/apriltag.py)

This issue has been met first when I used "pip install apriltag", it can be installed, but had this issue when I ran the code.

mkrogius, 2020-10-08:
Based on the path in the error message, I think your code is still trying to import the version of apriltag from pip (the path ends in apriltag.py instead of apriltag.cpython-37m-darwin.so). Can you try 'pip uninstall apriltag'? And if that doesn't work then you can try deleting the file: /Applications/anaconda3/lib/python3.7/site-packages/apriltag.py

florakth, 2020-10-08:
Thank you!  pip uninstalled  apriltag works, and I reinstall it as in the previous steps. But there is one note at last:

-- Installing: /usr/local/share/apriltag/cmake/apriltagTargets-release.cmake
-- Installing: /usr/local/share/apriltag/cmake/apriltagConfig.cmake
-- Installing: /usr/local/lib/pkgconfig/apriltag.pc
cp: /Users/flora/.local/lib/python3.7/site-packages: No such file or directory
-- Installing: /usr/local/bin/opencv_demo
-- Installing: /usr/local/bin/apriltag_demo

And when I import apriltag:
>>> from apriltag import apriltag
Segmentation fault: 11

mkrogius, 2020-10-08:
The line 'cp: /Users/flora/.local/lib/python3.7/site-packages: No such file or directory' makes me think that the library was still not installed correctly because python is trying to install the file to the wrong location. Lets try one thing, can you copy the file apriltag.cpython-37m-darwin.so to the same folder that your python script is in and then try running your script again?

florakth, 2020-10-08:
Hi, I tried to copy apriltag.cpython-37m-darwin.so to the same folder and also copied my script to the apriltag folder,
both has the same errors: Segmentation fault: 11

mkrogius, 2020-10-08:
Can you try adding the following lines to your script before the line where you import apriltag? It should hopefully cause a more useful error message to be displayed.

import faulthandler
faulthandler.enable()

mkrogius, 2020-10-08:
Also, can you run 'file apriltag.cpython-37m-darwin.so' in the terminal in any directory which contains the apriltag.cpython-37m-darwin.so file and paste the output here?

florakth, 2020-10-09:
Yes, I did it as you said, it shows:

(base)...$ python3 detect_marker.py
Fatal Python error: Segmentation fault
Current thread 0x00000001141335c0 (most recent call first):
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1043 in create_module
  File "<frozen importlib._bootstrap>", line 583 in module_from_spec
  File "<frozen importlib._bootstrap>", line 670 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 967 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
  File "detect_marker.py", line 7 in <module>
Segmentation fault: 11

mkrogius, 2020-10-09:
Can you run this command in the apriltag directory?:

file apriltag.cpython-37m-darwin.so

florakth, 2020-10-09:
No, it can not be run directly, it reminds me to choose an app to open it.

mkrogius, 2020-10-09:
I think there might be a slight misunderstanding here, I want you to type:
```
file apriltag.cpython-37m-darwin.so
```
on the command line, not just:
```
apriltag.cpython-37m-darwin.so
```

The command `file` is a unix program that will print out information about the contents of apriltag.cpython-37m-darwin.so.

florakth, 2020-10-09:
Sorry for that, it can be run:

(base) Floras-MacBook-Pro:apriltag flora$ file apriltag.cpython-37m-darwin.so
apriltag.cpython-37m-darwin.so: Mach-O 64-bit bundle x86_64

mkrogius, 2020-10-09:
Hmm this is getting trickier to debug. Do you have multiple versions of python installed? Do you use anaconda to manage your python environments?

florakth, 2020-10-09:
Yes, I use anaconda, the terminal will automatically come with (base), and I use python 3.7 but may have installed 2.7 too together with other libaries.

mkrogius, 2020-10-09:
Ok I think I'm starting to understand the problem. In the apriltag folder, can you run the following commands and tell me the output for each of them?

```
python3 python_build_flags.py
```

and 

```
make clean
make VERBOSE=1
```

florakth, 2020-10-09:
It is like this:

(base) Floras-MacBook-Pro:apriltag flora$ python3 python_build_flags.py
CFLAGS;-Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -Wall -Wstrict-prototypes -I/Applications/anaconda3/include -arch x86_64 -I/Applications/anaconda3/include -arch x86_64 -I/Applications/anaconda3/include/python3.7m -I/Applications/anaconda3/lib/python3.7/site-packages/numpy/core/include -Wno-strict-prototypes;LINKER;gcc;LDFLAGS;-bundle -undefined dynamic_lookup -L/Applications/anaconda3/lib -arch x86_64 -L/Applications/anaconda3/lib -arch x86_64 -L. -lpython3.7m -L/Applications/anaconda3/lib  -arch x86_64 -L/Applications/anaconda3/lib  -arch x86_64;EXT_SUFFIX;.cpython-37m-darwin.so;

(base) Floras-MacBook-Pro:apriltag flora$  make clean
(base) Floras-MacBook-Pro:apriltag flora$ make VERBOSE=1
/Applications/anaconda3/bin/cmake -S/Users/flora/desktop/KU/RPLproject/segmentation/apriltag -B/Users/flora/desktop/KU/RPLproject/segmentation/apriltag --check-build-system CMakeFiles/Makefile.cmake 0
/Applications/anaconda3/bin/cmake -E cmake_progress_start /Users/flora/desktop/KU/RPLproject/segmentation/apriltag/CMakeFiles /Users/flora/desktop/KU/RPLproject/segmentation/apriltag/CMakeFiles/progress.marks
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/Makefile2 all
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/apriltag.dir/build.make CMakeFiles/apriltag.dir/depend
cd /Users/flora/desktop/KU/RPLproject/segmentation/apriltag && /Applications/anaconda3/bin/cmake -E cmake_depends "Unix Makefiles" /Users/flora/desktop/KU/RPLproject/segmentation/apriltag /Users/flora/desktop/KU/RPLproject/segmentation/apriltag /Users/flora/desktop/KU/RPLproject/segmentation/apriltag /Users/flora/desktop/KU/RPLproject/segmentation/apriltag /Users/flora/desktop/KU/RPLproject/segmentation/apriltag/CMakeFiles/apriltag.dir/DependInfo.cmake --color=
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/apriltag.dir/build.make CMakeFiles/apriltag.dir/build
[  2%] Building C object CMakeFiles/apriltag.dir/apriltag.c.o
/Library/Developer/CommandLineTools/usr/bin/cc -Dapriltag_EXPORTS -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/. -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/common -I/apriltag  -O3 -DNDEBUG -fPIC   -o CMakeFiles/apriltag.dir/apriltag.c.o   -c /Users/flora/desktop/KU/RPLproject/segmentation/apriltag/apriltag.c
[  5%] Building C object CMakeFiles/apriltag.dir/apriltag_pose.c.o
/Library/Developer/CommandLineTools/usr/bin/cc -Dapriltag_EXPORTS -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/. -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/common -I/apriltag  -O3 -DNDEBUG -fPIC   -o CMakeFiles/apriltag.dir/apriltag_pose.c.o   -c /Users/flora/desktop/KU/RPLproject/segmentation/apriltag/apriltag_pose.c
[  7%] Building C object CMakeFiles/apriltag.dir/apriltag_quad_thresh.c.o
/Library/Developer/CommandLineTools/usr/bin/cc -Dapriltag_EXPORTS -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/. -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/common -I/apriltag  -O3 -DNDEBUG -fPIC   -o CMakeFiles/apriltag.dir/apriltag_quad_thresh.c.o   -c /Users/flora/desktop/KU/RPLproject/segmentation/apriltag/apriltag_quad_thresh.c
[ 10%] Building C object CMakeFiles/apriltag.dir/common/g2d.c.o
/Library/Developer/CommandLineTools/usr/bin/cc -Dapriltag_EXPORTS -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/. -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/common -I/apriltag  -O3 -DNDEBUG -fPIC   -o CMakeFiles/apriltag.dir/common/g2d.c.o   -c /Users/flora/desktop/KU/RPLproject/segmentation/apriltag/common/g2d.c
[ 12%] Building C object CMakeFiles/apriltag.dir/common/getopt.c.o
/Library/Developer/CommandLineTools/usr/bin/cc -Dapriltag_EXPORTS -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/. -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/common -I/apriltag  -O3 -DNDEBUG -fPIC   -o CMakeFiles/apriltag.dir/common/getopt.c.o   -c /Users/flora/desktop/KU/RPLproject/segmentation/apriltag/common/getopt.c
[ 15%] Building C object CMakeFiles/apriltag.dir/common/homography.c.o
/Library/Developer/CommandLineTools/usr/bin/cc -Dapriltag_EXPORTS -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/. -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/common -I/apriltag  -O3 -DNDEBUG -fPIC   -o CMakeFiles/apriltag.dir/common/homography.c.o   -c /Users/flora/desktop/KU/RPLproject/segmentation/apriltag/common/homography.c
[ 17%] Building C object CMakeFiles/apriltag.dir/common/image_u8.c.o
/Library/Developer/CommandLineTools/usr/bin/cc -Dapriltag_EXPORTS -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/. -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag -I/Users/flora/desktop/KU/RPLproject/segmentation/apriltag/common -I/apriltag  -O3 -DNDEBUG -fPIC   -o CMakeFiles/apriltag.dir/common/image_u8.c.o   -c /Users/flora/desktop/KU/RPLproject/segmentation/apriltag/common/image_u8.c
[ 20%] Building C object CMakeFiles/apriltag.dir/common/image_u8x3.c.o
[... cut ...] 3.4.7.dylib /Applications/anaconda3/lib/libopencv_features2d.3.4.7.dylib /Applications/anaconda3/lib/libopencv_flann.3.4.7.dylib /Applications/anaconda3/lib/libopencv_photo.3.4.7.dylib /Applications/anaconda3/lib/libopencv_imgproc.3.4.7.dylib /Applications/anaconda3/lib/libopencv_core.3.4.7.dylib 
[100%] Built target opencv_demo
/Applications/anaconda3/bin/cmake -E cmake_progress_start /Users/flora/desktop/KU/RPLproject/segmentation/apriltag/CMakeFiles 0
(base) Floras-MacBook-Pro:apriltag flora$

mkrogius, 2020-10-09:
Ok, looks like its not a conda issue. Please run the following at the command line:
```
gdb --args python3 detect_marker.py
```

This will take you into the gdb command line. Enter the following on the command line:
```
run
```
This will start the program. It should crash as before, but you should still be at the gdb command line. Enter the following:
```
backtrace
```
Now exit gdb by typing the following:
```
gdb
```
Please include all the output from these commands

florakth, 2020-10-09:
It seems no gdb...
(base) Floras-MacBook-Pro:apriltag flora$ gdb --args python3 detect_marker.py
-bash: gdb: command not found

florakth, 2020-10-09:
I tried to install gdb,  then get the results:

(base) Floras-MacBook-Pro:segmentation flora$ gdb --args python3 detect_marker.py
GNU gdb (GDB) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-apple-darwin18.7.0".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python3...

warning: `/tmp/lto.o': can't open to read symbols: No such file or directory.
(No debugging symbols found in python3)
(gdb) run
Starting program: /Applications/anaconda3/bin/python3 detect_marker.py
Note: this version of macOS has System Integrity Protection.
Because `startup-with-shell' is enabled, gdb has worked around this by
caching a copy of your shell.  The shell used by "run" is now:
    /Users/flora/Library/Caches/gdb/bin/bash
Unable to find Mach task port for process-id 5578: (os/kern) failure (0x5).
 (please check gdb is codesigned - see taskgated(8))
(gdb) backtrace
No stack.
(gdb) gdb
Undefined command: "gdb".  Try "help".
(gdb)

mkrogius, 2020-10-09:
It looks like gdb didn't work quite right because it says "Unable to find Mach task port for process-id 5578: (os/kern) failure (0x5)."

Can you try these instructions in order to get gdb to work? https://gist.github.com/hlissner/898b7dfc0a3b63824a70e15cd0180154

florakth, 2020-10-09:
Sorry, I tried two ways
 https://gist.github.com/hlissner/898b7dfc0a3b63824a70e15cd0180154
https://gist.github.com/gravitylow/fb595186ce6068537a6e9da6d8b5b96d
Both failed at the last few steps. 

Sometime I can not find the file mentioned in the steps have to google a lot of things.
I have to stop a little. 
Thank you again.

closed 2021-06-15 by mkrogius