AprilRobotics/apriltag #113: Cannot decode image if I install this system in MacOS

1999kevin, 2020-11-06:
I try to build this system in my MacBook, and it seems that the building process is fine. I can also build the apriltag_demo in example directory. But when I try to use this demo to decode some images, it fails, saying that there is an unknown marker. Any advices about why this bug happen?

[image]

1999kevin, 2020-11-06:
I didn't make install the files to my system. Only use "cmake ." and use CMake to generate aprilTag_demo, which can be found in build directory after building process. And I download some images in tag36_11 family to test

1999kevin, 2020-11-06:
After I run "make install", same bug happens.

mkrogius, 2020-11-12:
Can you try with a png file or pnm file?

mkrogius, 2020-11-12:
*Sorry, pnm or pgm file

mkrogius, 2020-11-12:
Or just try a different jpeg, the problem is a jpeg decode issue

1999kevin, 2020-11-12:
I try another jpg file, which has larger size. There is a still unknown marker, but I can get its id. How could this happen?
[image]

1999kevin, 2020-11-12:
One more question: does this demo program, or more specifically, image_u8_create_from_pnm(), really support png file? From the code in Apriltag_demo, I think it only supports pnm and pgm file, but from user guide, this function can handle the png file input, as shown at the first line of below code.
[image]

mkrogius, 2020-11-12:
The word "marker" in the error message "pjepg error: Unknown marker ed at offset 006e" does not refer to a fiducial marker, but instead refers to a sequence of bits within the jpeg file which the code does not recognize.

The demo program does not support png.

1999kevin, 2020-11-12:
I got it. I tried to input 10*10 pixels files, which are downloaded directly from **apriltag-imgs**, to the program and it didn't recognize. Does the algorithm really address the problem about the single pixel wide of the tag boundaries, which is mentioned in paper, _AprilTag 2: Efficient and robust fiducial detection_?

mkrogius, 2020-11-12:
Yes it does, for the typical use case of this algorithm which is for images taken with a real camera and optical system. What this means is that the minimum tag size that you could realistically detect (taking blur into account) is a little bit larger than the rendered image you are using.

However since I was a little curious I did explore more and make a small change (relaxing a threshold on tag size) that lets the program detect even these minimum size tags. Pull the latest version of the code and try running apriltag_demo again but this time set refine_edges to 'false'. Also please use this test image as it does not have any jpeg decode errors and also has some extra background pixels so the border of the tag is not directly on the edge of the image.
[image]

closed 2020-11-12 by mkrogius