AprilRobotics/apriltag #120: No such file or directory when including apriltag/apriltag.h

asasine, 2021-01-06:
Hello! We're trying to link against the apriltag library using its `apriltag::apriltag` target but we're gettng and error whenever we `#include <apriltag/apriltag.h>` into one of our project files:

```
In file included from /home/adam/Project/src/my_project/src/MySource.cpp:1:0:
/home/adam/Project/src/my_project/include/MyHeader.h:2:10: fatal error: apriltag/apriltag.h: No such file or directory

 #include <apriltag/apriltag.h>
          ^~~~~~~~~~~~~~~~~~~~~
```

In the project's CMakeLists.txt, we add apriltag as a link library:

```cmake
find_package(apriltag REQUIRED)

add_library(${PROJECT_NAME} SHARED
  include/MyHeader.h
  src/MySrc.cpp
)

include(GNUInstallDirs)
target_include_directories(${PROJECT_NAME}
  PUBLIC
    "$<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include/>"
    "$<INSTALL_INTERFACE:${CMAKE_INSTALL_INCLUDEDIR}>/${PROJECT_NAME}"
)

target_link_libraries(${PROJECT_NAME} PUBLIC 
  apriltag::apriltag
) 
```

The apriltag library is cloned as a submodule alongside our project and it compiles successfully.

I added this block to our CMake file to check if the library and target were being found and and it prints during the build which indicates they were:

```cmake
if(${apriltag_FOUND} AND TARGET apriltag::apriltag)
  message("apriltag library and target found")
endif()
```

To unblock myself, I dug through the CMakeLists.txt for the apriltag library and noticed that when the shared library is installed and added to the `apriltagTargets` export, it does not include the `INCLUDES DESTINATION` option. After adding this option to my local clone, the original error went away, and my project compiled successfully.

```cmake
# install library
install(TARGETS ${PROJECT_NAME} EXPORT apriltagTargets
    LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR}
    ARCHIVE DESTINATION ${CMAKE_INSTALL_LIBDIR}

    # I added the below line
    INCLUDES DESTINATION ${CMAKE_INSTALL_INCLUDEDIR}
)
```

My apologies if this is not the proper use of export targets or the library as I'm still learning CMake. If this is the correct solution, I'd be happy to open a PR with the change.

anilharish, 2021-03-17:
I spent a day trying to debug this issue - this needs to be addressed a 

> Hello! We're trying to link against the apriltag library using its `apriltag::apriltag` target but we're gettng and error whenever we `#include <apriltag/apriltag.h>` into one of our project files: [...]

Thanks @asasine It took me almost a day to debug this and I think your addition is important. I too spent hours going though CMake error logs to fix this and was about to open an issue when I found your solution. Should have checked here first 😓 - keeping my rant apart that is a much needed bug fix !

mkrogius, 2021-03-17:
Can you paste the output you get from running `sudo make install` as per the install instructions?

Also, can you try `#include "apriltag.h"` instead of your include line?

QuentinTorg, 2021-04-28:
I think #140 will solve this problem. 

It could also explain why this problem seems to exist in some projects but not others. Sometimes other libraries included in a project will add the `<install_prefix>/include` directory, which will cause `#include <apriltag/apriltag.h>` to become valid again

mkrogius, 2021-05-07:
Fixed by #140

closed 2021-05-07 by mkrogius